ansys/pyaedt #4255: Bug located in pyaedt\modules\report_templates.py

Acaptian, 2024-02-21:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Please, provides an example of the use of the "add_limit_line_from_points" or "add_limit_line_from_equation()" in Hfss3Dlayout.

### Steps To Reproduce

import pyaedt from pyaedt.modules import report_templates
import pyaedt

h3d = pyaedt.Hfss3dLayout(projectname=project_path, non_graphical=False, new_desktop_session=True)
RL = h3d.get_all_return_loss_list()
RL_dB = ["dB({})".format(item) for item in RL]
h3d.post.create_report(expressions=RL_dB, plotname="Return Loss")

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

cffi            1.16.0  
clr-loader      0.2.6   
pip             21.1.2  
plumbum         1.8.2   
psutil          5.9.7   
pyaedt          0.7.11   
pycparser       2.21
pywin32         306
rpyc            5.3.1
setuptools      57.0.0
wheel           0.36.2

gmalinve, 2024-02-22:
Hi @Acaptian,

We currently don't have any example available for Hfss3dlayout but we do have it for a Circuit example.
Please look at it here: https://aedt.docs.pyansys.com/version/stable/examples/07-Circuit/Circuit_Transient.html#create-report-inside-aedt.
Bear in mind that this method only works in graphical mode.

Hope is useful somehow!
Kind regards,

Giulia

gmalinve, 2024-03-06:
Hi @Acaptian,

was my answer helpful? 
please do let me know so we can close this issue.

Thanks,

Giulia

Acaptian, 2024-03-08:
Hi [gmalinve](https://github.com/gmalinve)
Thank you for the Circuit example，but it dosen't show how to use the  "add_limit_line_from_points" or "add_limit_line_from_equation()"

Samuelopez-ansys, 2024-03-08:
@Acaptian Yes it does:

new_report.add_limit_line_from_points([60, 80], [1, 1], "ns", "V")

gmalinve, 2024-03-13:
Hi @Acaptian,

did you check the example I suggested?
Please give us a feedback so we can close this issue in case it replies to your question.

Thanks,

Giulia

closed 2024-03-13 by gmalinve