ansys/pyaedt #4270: Bug located in copy

siva-krishnaswamy, 2024-02-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

copying was not working when an single object was selected. The issue here is that since the selection comes as an list but native API needs only name if there is single object selection

### Steps To Reproduce

ipk.modeler.copy(single_object)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

aiohttp==3.8.3
aiosignal==1.3.1
appdirs==1.4.4
async-timeout==4.0.2
attrs==23.2.0
certifi==2022.12.7
cffi==1.15.1
cfgv==3.3.1
charset-normalizer==2.1.1
clr-loader==0.2.4
colorama==0.4.6
contourpy==1.0.6
cycler==0.11.0
distlib==0.3.6
exceptiongroup==1.1.1
filelock==3.8.2
fonttools==4.38.0
frozenlist==1.3.3
identify==2.5.9
idna==3.4
imageio==2.22.4
iniconfig==2.0.0
jsonschema==4.20.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.4
matplotlib==3.6.2
multidict==6.0.3
nodeenv==1.7.0
numpy==1.23.5
packaging==22.0
pandas==1.5.2
Pillow==9.3.0
platformdirs==2.6.0
pluggy==1.0.0
plumbum==1.8.0
pooch==1.6.0
pre-commit==3.6.0
psutil==5.9.4
-e git+https://github.com/pyansys/pyaedt.git@d2a299792c6597ba8ab0301cc1961dd8740a8140#egg=pyaedt
pycparser==2.21
pyparsing==3.0.9
pytest==7.3.1
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.6
pyvista==0.37.0
pywin32==305
PyYAML==6.0
referencing==0.32.1
requests==2.28.1
rpds-py==0.16.2
rpyc==5.0.1
scooby==0.7.0
six==1.16.0
toml==0.10.2
tomli==2.0.1
urllib3==1.26.13
virtualenv==20.17.1
vtk==9.2.2
wslink==1.9.2
yarl==1.8.2

Samuelopez-ansys, 2024-02-23:
@siva-krishnaswamy self.convert_to_selections(object_list) should return a string if you pass one element.

This method returns:
",".join([str(i) for i in objnames])

I am not able to reproduce the issue in 2023R2 and Python 3.10.

siva-krishnaswamy, 2024-02-23:
@Samuelopez-ansys ,I was trying in 2024R2 and noticed this issue. I will check in 2023R2 and also in 2024R2. I will also add an unit test for one object copy. I will get back to you. Thanks

Regards,
Siva

siva-krishnaswamy, 2024-02-23:
@Samuelopez-ansys, Sorry it was working fine. I will close this issue

closed 2024-02-23 by siva-krishnaswamy