ansys/pyaedt #4283: Doc pyaedt failure due to incorrect GRPC port connection

Nanaaaaaaa19, 2024-02-26:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

While trying to generate a documentation of my own scripts locally with Sphinx, I encountered some issues to generate them due to a connection to the wrong GRPC port while opening AEDT. As you will see on the screenshots below, my scripts are all failing when executed through the doc, if ever you are experimenting something similar, I would recommend you to put that following line at the beginning of your script that will disable the GRPC com with AEDT however the doc will work fine like that: `pyaedt.settings.use_grpc_api = False` 

### Steps To Reproduce

 write whichever pyaedt script that creates something from scratch ( but you may never encounter that issue if you are lucky )
[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

accessible-pygments==0.0.4
aiohttp==3.9.3
aiosignal==1.3.1
alabaster==0.7.16
ansys-pythonnet==3.1.0rc3
ansys-sphinx-theme==0.13.1
anyio==4.3.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
Babel==2.14.0
backcall==0.2.0
beautifulsoup4==4.12.3
bleach==6.1.0
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.1
commonmark==0.9.1
contourpy==1.2.0
cycler==0.12.1
debugpy==1.8.1
decorator==5.1.1
defusedxml==0.7.1
docutils==0.20.1
et-xmlfile==1.1.0
executing==2.0.1
fastjsonschema==2.19.1
fiona==1.9.5
fonttools==4.49.0
fpdf2==2.7.8
fqdn==1.5.1
frozenlist==1.4.1
geopandas==0.14.3
idna==3.6
imageio==2.31.5
imageio-ffmpeg==0.4.9
imagesize==1.4.1
importlib-metadata==7.0.1
iniconfig==2.0.0
ipykernel==6.29.2
ipython==8.16.1
ipywidgets==8.1.1
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.3
joblib==1.3.2
json5==0.9.17
jsonpointer==2.4
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
jupyter-events==0.9.0
jupyter-lsp==2.2.2
jupyter_client==8.6.0
jupyter_core==5.7.1
jupyter_server==2.12.5
jupyter_server_terminals==0.5.2
jupyterlab==4.0.6
jupyterlab_pygments==0.3.0
jupyterlab_server==2.25.3
jupyterlab_widgets==3.0.10
kiwisolver==1.4.5
livereload==2.6.3
MarkupSafe==2.1.5
matplotlib==3.8.0
matplotlib-inline==0.1.6
mistune==3.0.2
more-itertools==10.2.0
multidict==6.0.5
nbclient==0.9.0
nbconvert==7.16.1
nbformat==5.9.2
nbsphinx==0.9.3
nest-asyncio==1.6.0
networkx==3.2.1
notebook_shim==0.2.4
numpy==1.26.0
numpydoc==1.6.0
openpyxl==3.1.2
osmnx==1.9.1
overrides==7.7.0
packaging==23.2
pandas==2.1.1
pandocfilters==1.5.1
parso==0.8.3
pickleshare==0.7.5
pillow==10.2.0
platformdirs==4.2.0
pluggy==1.4.0
plumbum==1.8.2
pooch==1.8.1
prometheus_client==0.20.0
prompt-toolkit==3.0.43
psutil==5.9.8
pure-eval==0.2.2
pyaedt==0.7.12
pycparser==2.21
pydata-sphinx-theme==0.14.4
Pygments==2.17.2
pypandoc==1.11
pyparsing==3.1.1
pyproj==3.6.1
pytest==8.0.1
pytest-sphinx==0.5.0
python-dateutil==2.8.2
python-json-logger==2.0.7
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.42.2
pywin32==306
pywinpty==2.0.12
PyYAML==6.0.1
pyzmq==25.1.2
recommonmark==0.7.1
referencing==0.33.0
requests==2.31.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.18.0
rpyc==5.3.1
scikit-learn==1.3.1
scikit-rf==0.31.0
scipy==1.12.0
scooby==0.9.2
Send2Trash==1.8.2
shapely==2.0.3
six==1.16.0
sniffio==1.3.0
snowballstemmer==2.2.0
soupsieve==2.5
Sphinx==7.2.6
sphinx-autobuild==2021.3.14
sphinx-autodoc-typehints==1.24.0
sphinx-copybutton==0.5.2
sphinx-gallery==0.14.0
sphinx-jinja==2.0.2
sphinx-notfound-page==1.0.0
sphinx_design==0.5.0
sphinxcontrib-applehelp==1.0.8
sphinxcontrib-devhelp==1.0.6
sphinxcontrib-htmlhelp==2.0.5
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.7
sphinxcontrib-serializinghtml==1.1.10
SRTM.py==0.3.7
stack-data==0.6.3
tabulate==0.9.0
terminado==0.18.0
threadpoolctl==3.3.0
tinycss2==1.2.1
tornado==6.4
traitlets==5.14.1
trame==3.5.2
trame-client==2.16.1
trame-server==2.17.2
trame-vtk==2.8.0
trame-vuetify==2.4.2
types-python-dateutil==2.8.19.20240106
typing_extensions==4.9.0
tzdata==2024.1
uri-template==1.3.0
urllib3==2.2.1
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.13
webcolors==1.13
webencodings==0.5.1
websocket-client==1.7.0
widgetsnbextension==4.0.10
wslink==1.12.4
yarl==1.9.4
zipp==3.17.0

Samuelopez-ansys, 2024-02-26:
@Nanaaaaaaa19 I think it is not a PyAEDT issue, it is related to AEDT GRPC, and for some reason your computer does not accept the port communication. If can reproduce it, you could submit it as an internal bug in Ansys to fix it. 

I will close the issue. If you can reproduce the issue with a simple script, we could also analyze it together. Thanks

SMoraisAnsys, 2024-02-28:
> @Nanaaaaaaa19 I think it is not a PyAEDT issue, it is related to AEDT GRPC, and for some reason your computer does not accept the port communication. If can reproduce it, you could submit it as an [...]

@Samuelopez-ansys I was the one to ask @Nanaaaaaaa19 to open an issue. I'll talk with her about creating a reproducer and providing guidance to our customers when such problem arises. Since you said you would close it and didn't, I won't close it but feel free to do so. If we are able to create a reproducer, we'll reopen it and work on code changes to notify users about the problem.

SMoraisAnsys, 2024-03-11:
@Samuelopez-ansys Closing this issue for the moment as we are not able to reproduce the bug.

As a reminder for future reads, @Nanaaaaaaa19 had a problem when her examples were running through documentation build (she leveras make.bat file) see https://github.com/ansys/pyaedt/issues/4283#issue-2154483927. However, when she executed the examples directly, the bug was not happening ! To make things work through the documentation, she had to disable the use of grpc.

closed 2024-03-11 by SMoraisAnsys