ansys/pyaedt #4286: LSF submission, Bug located in desktop.py (pyaedt-0.7.12)

kruglovIFX, 2024-02-27:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

desktop.py line 1912 missing an "=" sign:
`elif "NumCores" in line:`
should be
`elif "NumCores=" in line:`
otherwise submission to LSF doesn't work

### Steps To Reproduce

Execute test.submit_job() on LSF

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

Does not depend on other packages. Same issue with other 0.6.XX and 0.7.XX versions

kruglovIFX, 2024-02-27:
Error message in EDT if "=" sign not added:
[image]

Samuelopez-ansys, 2024-02-27:
Hi @kruglovIFX , PyAEDT is an open-source library so that next time you can create a pull request and it will be merged into PyAEDT if it passes all the tests and PyAnsys checks. I have created a pull request with your suggestions. 

https://github.com/ansys/pyaedt/pull/4290/files

closed 2024-02-27 by Samuelopez-ansys

kruglovIFX, 2024-02-27:
Hi @Samuelopez-ansys,
Alright, will do next time. Thanks!

PR #4290 (2024-02-27): Add custom LSF
Close #4286 #4288 #4287