ansys/pyaedt #4300: BUG: PyAEDT API Maxwell 2D parametrics variable unit issue in 24R1

leoleo999, 2024-02-28:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hello engineers, 

I am runing the example named 'Maxwell2D_DCConduction'. 
In the 'parametric sweep' section, the sweeping variable 'materialindex' induced an error showing that the unit of this variable is different from the variable definition. However, this 'materialindex' is an integer which does not have any unit in the definition. I tried 'str()' to set the sweeping values, but it did not work. Other variables with units work fine for the 'parametrics' API. 

I attached the code and error message. 
Look forward to your reply.

Best regards,
Leo Liu
[image]
[Error_code.txt](https://github.com/ansys/pyaedt/files/14439042/Error_code.txt)

### Steps To Reproduce

Run the attached error code in Maxwell 24R1 ->Tools ->Toolkit ->PersonalLib ->PyAEDT ->Run PyAEDT Script

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

anyio==4.2.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
Babel==2.14.0
beautifulsoup4==4.12.3
bleach==6.1.0
certifi==2023.11.17
cffi==1.16.0
charset-normalizer==3.3.2
colorama==0.4.6
comm==0.2.1
debugpy==1.8.0
decorator==5.1.1
defusedxml==0.7.1
executing==2.0.1
fastjsonschema==2.19.1
fqdn==1.5.1
idna==3.6
ipykernel==6.29.0
ipython==8.20.0
ipywidgets==8.1.1
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.3
json5==0.9.14
jsonpointer==2.4
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
jupyter==1.0.0
jupyter-console==6.6.3
jupyter-events==0.9.0
jupyter-lsp==2.2.2
jupyter_client==8.6.0
jupyter_core==5.7.1
jupyter_server==2.12.5
jupyter_server_terminals==0.5.2
jupyterlab==4.0.11
jupyterlab-widgets==3.0.9
jupyterlab_pygments==0.3.0
jupyterlab_server==2.25.2
MarkupSafe==2.1.4
matplotlib-inline==0.1.6
mistune==3.0.2
nbclient==0.9.0
nbconvert==7.14.2
nbformat==5.9.2
nest-asyncio==1.6.0
notebook==7.0.7
notebook_shim==0.2.3
overrides==7.6.0
packaging==23.2
pandocfilters==1.5.1
parso==0.8.3
platformdirs==4.1.0
prometheus-client==0.19.0
prompt-toolkit==3.0.43
psutil==5.9.8
pure-eval==0.2.2
pycparser==2.21
Pygments==2.17.2
python-dateutil==2.8.2
python-json-logger==2.0.7
pywin32==306
pywinpty==2.0.12
PyYAML==6.0.1
pyzmq==25.1.2
qtconsole==5.5.1
QtPy==2.4.1
referencing==0.32.1
requests==2.31.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.17.1
Send2Trash==1.8.2
six==1.16.0
sniffio==1.3.0
soupsieve==2.5
stack-data==0.6.3
terminado==0.18.0
tinycss2==1.2.1
tornado==6.4
traitlets==5.14.1
types-python-dateutil==2.8.19.20240106
uri-template==1.3.0
urllib3==2.1.0
wcwidth==0.2.13
webcolors==1.13
webencodings==0.5.1
websocket-client==1.7.0
widgetsnbextension==4.0.9

gmalinve, 2024-03-05:
Hi @leoleo999,

I tested the example with 2024.1 and couldn't reproduce your issue, it works as expected.
What I notice from your environment is that pyaedt is not listed, maybe you didn't use pip freeze command within your virtual environment. 
Please check the pyaedt version you have in your venv.

Hope this helps,

Giulia

leoleo999, 2024-03-05:
Hi Giulia,

Thank you so much for the help. Sorry I do not know the pip freeze because I know little about python. Could guide me how to insert this pip freeze with some tutorials ?
I just learned the pyaedt by running those Maxwell examples with APIs online in our website.
In addition, I was trying to reproduce the issue for you, but seems the pyAEDT GUI has been changed as shown below:
***@***.***

I remember there is one more option named ‘Run RpyAEDT Script’ between the ‘Jupter’ and ‘Run_PyAEDT_Script’. And it navigates me to the code file location. But now it disappeared after I install the PyAEDT.
Is there any tutorial for this updated difference ? How can I run the downloaded examples.

Sorry for these stupid questions.
Look forward to your reply.
Best regards,
Leo Liu

[External Sender]

leoleo999, 2024-03-05:
Hi Giulia,

I re-installed and updated the PyAEDT , and the issue was resolved! The API works too.

Amazing.
Thank you so much.
Leo Liu