ansys/pyaedt #4304: Unable to deleted all unclassified objects

jmaravin, 2024-02-29:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am unable to delete all unclassified objects through the following commands: 
    unclass = m3d.modeler.unclassified_objects
    m3d.modeler.delete(unclass)

For some reason, the above code deletes half of the unclassified objects so I need to run it several times. However, when there´s only one unclassified object left, the code does not delete it. 

### Steps To Reproduce

Run the following command in any project with unclassified objects and see if it delets all of these unclassified objects. 

for i in range(1, 10):
    unclass = m3d.modeler.unclassified_objects
    m3d.modeler.delete(unclass)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-pythonnet==3.1.0rc3
attrs==23.2.0
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.2.0
customtkinter==5.2.2
cycler==0.12.1
darkdetect==0.8.0
defusedxml==0.7.1
et-xmlfile==1.1.0
fiona==1.9.5
fonttools==4.49.0
fpdf2==2.7.8
geopandas==0.14.3
idna==3.6
imageio==2.34.0
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
matplotlib==3.8.0
networkx==3.2.1
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.9.1
packaging==23.2
pandas==2.1.1
pillow==10.2.0
platformdirs==4.2.0
plumbum==1.8.2
pooch==1.8.0
psutil==5.9.8
pyaedt==0.8.1
pycparser==2.21
pyedb==0.4.0
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.42.2
pywin32==306
referencing==0.33.0
requests==2.31.0
rpds-py==0.18.0
rpyc==5.3.1
scikit-rf==0.31.0
scipy==1.12.0
scooby==0.9.2
shapely==2.0.3
six==1.16.0
SRTM.py==0.3.7
tzdata==2024.1
urllib3==2.2.0
utm==0.7.0
vtk==9.2.6

Samuelopez-ansys, 2024-02-29:
@jmaravin Please provide an AEDT project to reproduce the issue.

gmalinve, 2024-03-11:
Hi @jmaravin,

Your issue has been tackled in this PR #4349 and most probably will be available in the next release.

Kind regards,

Giulia

jmaravin, 2024-03-11:
Great. Thanks @gmalinve appreciate it
Looking forward to it

closed 2024-03-11 by gmalinve