ansys/pyaedt #4373: Bug located in ...

mikevacany, 2024-03-18:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hello,

I am trying to use pyAEDT to script the DDR IBIS model scenario creation. I started with this example: https://aedt.docs.pyansys.com/version/0.7/examples/07-Circuit/Circuit_Transient.html. However, I am facing issue once I want to change IBIS model instance property 'power' from 'internal' to 'external'. This can be done manually from EDT GUI but not from pyAEDT. I attach a screenshot showing errors if two different approach to change the parameter are used. Other properties may be changed from pyAEDT. Even the 'power' property may be changed, but only 'external' -> 'internal' direction.

Can you confirm a bug or give an advice how to proceed? Customer service redirected me here.

BR,
Michael

[image]

### Steps To Reproduce

In the screenshot.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

Running directly from EDT pyAEDT console:

```
Python 3.10.9 (remotes/origin/4e018266dafe29ec2836492d71d245fcfd366d07-dirty:4e018266, Jan  3 2) [MSC v.1920 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.22.2 -- An enhanced Interactive Python. Type '?' for help.
Loading the PyAEDT Console.
Initializing new desktop!
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Log on file is enabled
PyAEDT INFO: Log on Desktop Message Manager is enabled
PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
PyAEDT INFO: {14084: 50051}
PyAEDT INFO: Launching PyAEDT outside AEDT with gRPC plugin.
PyAEDT INFO: Launching AEDT using the gRPC plugin.
PyAEDT INFO: Connecting to AEDT session on gRPC port 50051
PyAEDT INFO: AEDT installation Path C:\Program Files\AnsysEM\v241\Win64
PyAEDT INFO: pyaedt v0.7.0
PyAEDT INFO: Python version 3.10.9 (remotes/origin/4e018266dafe29ec2836492d71d245fcfd366d07-dirty:4e018266, Jan  3 2) [MSC v.1920 64 bit (AMD64)]
PyAEDT INFO: AEDT 2024.1.0 Build Date 2023-11-27 22:16:18
```

mikevacany, 2024-03-19:
@Samuelopez-ansys, could a workaround be used here similar as you proposed in https://github.com/ansys/pyaedt/discussions/4374? So, I could move forward.

Samuelopez-ansys, 2024-03-19:
@mikevacany Ok great, if it is something that you would like to have in Pyaedt, please create an issue. And contributors will check it.

mikevacany, 2024-03-19:
@Samuelopez-ansys, this bug is not resolved, I am just asking if there is a similar workaround, you provided in the other issue.

mikevacany, 2024-03-19:
@Samuelopez-ansys, can you reopen this issue?

Samuelopez-ansys, 2024-03-20:
Hi @mikevacany ,

PyAEDT is an open-source code, if you know how to fix the issue, you could try :)

If you do not have time or you can not fix it, one contributor from the community will help, but there are other Issues in the backlog. Thanks

gkorompi, 2025-06-24:
@mikevacany  @Samuelopez-ansys  I will have a look. Thanks guys