ansys/pyaedt #4375: materials.add_surface_material does not set emissivity in AEDT, and mat.update() does not update it in AEDT either

clguenthe, 2024-03-18:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The following code will create a SurfaceMaterial in pyaedt and AEDT, but the emissivity value will be only visible inside pyaedt and is not transferred correctly to AEDT:
```
from pyaedt import Icepak
icepak = Icepak()
mat = icepak.materials.add_surface_material("test", 10)

```
AEDT: 
[image]

There is a log message ``PyAEDT INFO: Material has been added. Edit it to update in Desktop.`` but it is unclear what "edit" refers to in this case (since my material definition is already complete, I don't want to edit it any further), and it is also not mentioned in the documentation of this method that any further step is required to update the material definition inside AEDT.

An explicit call to SurfaceMaterial.update() also does not update the emissivity:
```
>>> mat.update()
True
```
AEDT:
[image]

The underlying problem seems to be that the `SurfaceMaterial.update()` method uses `self._get_args()` to determine the arguments to be passed to `self.odefinition_manager.EditSurfaceMaterial(self.name, args)`, which does not yield the emissivity:
```
>>> mat._get_args()
['NAME:test', 'CoordinateSystemType:=', 'Cartesian', 'BulkOrSurfaceType:=', 1, ['NAME:PhysicsTypes', 'set:=', ['Thermal']]]
```
[image]

As a workaround, I can set the emissivity manually through the property setter. This works and the emissivity value is updated in AEDT. Furthermore, since the property setter explicitly calls
`self._update_props("surface_emissivity", value)`
it makes the `surface_emissivity` appear in the list of arguments returned by ``_get_args()``:
```
>>> mat.emissivity = 10
>>> mat._get_args()
['NAME:test2', 'CoordinateSystemType:=', 'Cartesian', 'BulkOrSurfaceType:=', 1, ['NAME:PhysicsTypes', 'set:=', ['Thermal']], 'surface_emissivity:=', '10']
```

### Steps To Reproduce

see above.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

ansys-pythonnet==3.1.0rc3
attrs==23.2.0
cffi==1.16.0
clr-loader==0.2.6
defusedxml==0.7.1
fonttools==4.50.0
fpdf2==2.7.8
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
pillow==10.2.0
plumbum==1.8.2
psutil==5.9.8
pyaedt==0.8.2
pycparser==2.21
pyedb==0.5.2
pytomlpp==1.0.13
pywin32==306
referencing==0.34.0
rpds-py==0.18.0
rpyc==6.0.0

clguenthe, 2024-03-20:
FYI: I was able to track this issue back to the changes in this PR: https://github.com/ansys/pyaedt/commit/e44a1cd9fc06cf830fe1329cb6392215d47cb390
-> before that, the emissivity value was set and transferred to AEDT directly when it was set, since the update_material flag was True.

lorenzovecchietti, 2024-10-10:
Hi @clguenthe, thanks for point out the issue. I think this was solved during the summer by @Samuelopez-ansys as I'm not able to reproduce the issue. Running `mat = icepak.materials.add_surface_material("test", 10)` even without update creates the material properly.

But I agree that the info message is confusing, so I created the mentioned PR.

PR #5274 (2024-10-11): FIX: Remove confusing/erroneous info message
Close #4375
closed 2024-10-11 by Samuelopez-ansys