ansys/pyaedt #4380: Design creation is very slow

mon-kev, 2024-03-19:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Launching a new Maxwell2D or Maxwell3D desktop session is very slow.

There is an initial ~10 seconds delay when [Desktop](https://github.com/ansys/pyaedt/blob/224d99353b2084935ade9cf91a3626afc6bc255d/pyaedt/desktop.py#L390) is initialised. And it takes ~1 min to execute [this step](https://github.com/ansys/pyaedt/blob/224d99353b2084935ade9cf91a3626afc6bc255d/pyaedt/application/Design.py#L260) during the `Design` initialisition. 

I am running pyaedt on a Windows machine inside a Ubuntu 22 docker container. I have noticed that the `Design` initialisation step is faster (~10-20 s) when I run it on Windows, in which case AEDT is launched with the COM plugin instead of gRPC. I have also noticed that it makes a difference which AEDT version I use. On Windows, the design creation takes roughly 2 times longer with v2024.1 than with v2022.2.

View log below:
```
Initializing new desktop!
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Log on file is enabled
PyAEDT INFO: Log on Desktop Message Manager is enabled
PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
PyAEDT INFO: Launching PyAEDT outside AEDT with gRPC plugin.
PyAEDT INFO: Launching AEDT using the gRPC plugin.
PyAEDT INFO: New AEDT session is starting on gRPC port 48679
/usr/local/lib/python3.11/subprocess.py:1127: ResourceWarning: subprocess 34868 is still running
  _warn("subprocess %s is still running" % self.pid,
ResourceWarning: Enable tracemalloc to get the object allocation traceback
PyAEDT INFO: AEDT installation Path /opt/AnsysEM/v241/Linux64
PyAEDT INFO: Ansoft.ElectronicsDesktop.2024.1 version started with process ID 35186.
PyAEDT INFO: pyaedt v0.7.12
PyAEDT INFO: Python version 3.11.7 (main, Jan 18 2024, 02:54:56) [GCC 11.4.0]
PyAEDT INFO: AEDT 2024.1.0 Build Date 2023-11-19 22:05:20
PyAEDT INFO: Project project_name has been created.
PyAEDT INFO: Added design 'design_name' of type Maxwell 2D.
```

Is there any way to speed the initial design creation up?

### Steps To Reproduce

Execute the following:

```python
import pyaedt
with pyaedt.Maxwell2d(
    new_desktop_session=True,
    non_graphical=True,
    close_on_exit=True,
    project_name="project_name",
    design_name="design_name",
) as maxwell:
    pass
  ```

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

about-time==4.2.1
adaptive==1.1.0
aiofiles==23.2.1
alive-progress==3.1.5
ansys-pythonnet==3.1.0rc3
anyio==4.3.0
application_properties==0.8.2
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
autograd==1.6.2
Babel==2.14.0
beautifulsoup4==4.12.3
bleach==6.1.0
boltons==23.1.1
build==1.1.1
CacheControl==0.14.0
certifi==2024.2.2
cffi==1.16.0
cfgv==3.4.0
charset-normalizer==3.3.2
cleo==2.1.0
cloudpickle==3.0.0
clr-loader==0.2.6
cma==3.2.2
Columnar==1.4.1
comm==0.2.1
contourpy==1.2.0
coverage==7.4.3
crashtest==0.4.1
cryptography==42.0.5
cycler==0.12.1
ddt==1.7.2
debugpy==1.8.1
decorator==5.1.1
defusedxml==0.7.1
Deprecated==1.2.14
dill==0.3.8
distlib==0.3.8
distro==1.9.0
dnspython==2.6.1
dotnetcore2==3.1.23
dulwich==0.21.7
execnet==2.0.2
executing==2.0.1
ezdxf==1.2.0
fastjsonschema==2.19.1
filelock==3.13.1
fonttools==4.49.0
fpdf2==2.7.8
fqdn==1.5.1
frozendict==2.3.10
future==1.0.0
geomdl==5.3.1
gmsh==4.12.2
grapheme==0.6.0
h11==0.14.0
h5py==3.10.0
httpcore==1.0.4
httpx==0.27.0
identify==2.5.35
idna==3.6
imageio==2.34.0
importlib-metadata==7.0.1
iniconfig==2.0.0
installer==0.7.0
ipykernel==6.29.3
ipynbname==2023.2.0.0
ipython==8.22.2
ipywidgets==8.1.2
isoduration==20.11.0
jaraco.classes==3.3.1
jedi==0.19.1
jeepney==0.8.0
Jinja2==3.1.3
joblib==1.3.2
json5==0.9.20
jsonpointer==2.4
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
junitparser==2.8.0
jupyter-events==0.9.0
jupyter-lsp==2.2.4
jupyter_client==8.6.0
jupyter_core==5.7.1
jupyter_server==2.13.0
jupyter_server_terminals==0.5.2
jupyterlab==4.1.3
jupyterlab_pygments==0.3.0
jupyterlab_server==2.25.3
jupyterlab_widgets==3.0.10
keyring==24.3.1
kiwisolver==1.4.5
loky==3.4.1
lxml==5.1.0
MarkupSafe==2.1.5
matplotlib==3.7.5
matplotlib-inline==0.1.6
meshio==4.4.6
mistune==3.0.2
more-itertools==9.1.0
mpmath==1.3.0
msgpack==1.0.8
nbclient==0.9.0
nbconvert==7.16.2
nbformat==5.9.2
nest-asyncio==1.6.0
networkx==3.1
nodeenv==1.8.0
notebook==7.1.1
notebook_shim==0.2.4
numpy==1.26.4
objectgmsh==0.12
overrides==7.7.0
packaging==23.2
pandas==2.2.1
pandocfilters==1.5.1
parso==0.8.3
pdoc==13.0.1
pexpect==4.9.0
pillow==10.2.0
pkginfo==1.10.0
platformdirs==4.2.0
pluggy==1.4.0
plumbum==1.8.2
poetry==1.8.2
poetry-core==1.9.0
poetry-plugin-export==1.6.0
pooch==1.8.1
pre-commit==3.6.2
prometheus_client==0.20.0
prompt-toolkit==3.0.43
psutil==5.9.8
ptyprocess==0.7.0
pure-eval==0.2.2
pyaedt==0.7.12
pycparser==2.21
pydot==1.4.2
Pygments==2.17.2
pymarkdownlnt==0.9.17
pymongo==4.6.2
pymoo==0.6.1.1
pyparsing==3.1.2
pyproject_hooks==1.0.0
pyqtgraph==0.13.3
pyright==1.1.351
pytest==7.4.4
pytest-cov==4.1.0
pytest-sugar==0.9.7
pytest-xdist==3.5.0
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytomlpp==1.0.13
pytz==2024.1
pyuff==2.4.2
pyvista==0.37.0
PyYAML==6.0.1
pyzmq==25.1.2
QtPy==2.4.1
rapidfuzz==3.6.2
referencing==0.33.0
requests==2.31.0
requests-toolbelt==1.0.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.18.0
rpyc==5.3.1
ruff==0.2.2
SciDataTool==2.5.0
scikit-learn==1.4.1.post1
scipy==1.12.0
scooby==0.9.2
SecretStorage==3.3.3
Send2Trash==1.8.2
shapely==2.0.3
shellingham==1.5.4
six==1.16.0
sniffio==1.3.1
sortedcollections==2.1.0
sortedcontainers==2.4.0
soupsieve==2.5
stack-data==0.6.3
sympy==1.12
termcolor==2.4.0
terminado==0.18.0
threadpoolctl==3.3.0
timeout-decorator==0.5.0
tinycss2==1.2.1
tomli==2.0.1
tomlkit==0.12.4
toolz==0.12.1
tornado==6.4
tqdm==4.66.2
traitlets==5.14.1
trove-classifiers==2024.3.3
types-python-dateutil==2.8.19.20240106
typing_extensions==4.10.0
tzdata==2024.1
unittest-xml-reporting==3.2.0
unyt==2.9.5
uri-template==1.3.0
urllib3==2.2.1
versioningit==3.0.0
virtualenv==20.25.1
vtk==9.3.0
wcwidth==0.2.13
webcolors==1.13
webencodings==0.5.1
websocket-client==1.7.0
widgetsnbextension==4.0.10
wrapt==1.16.0
xlrd==2.0.1
XlsxWriter==3.2.0
xlwt==1.3.0
zipp==3.17.0

mon-kev, 2024-03-21:
Launching AEDT manually works fine.
I updated pyaedt to v0.8.2, but the problem still occurs.

It seems weird to me, that it's not connecting to the initial session which is the slowest part, but inserting a design:
https://github.com/ansys/pyaedt/blob/224d99353b2084935ade9cf91a3626afc6bc255d/pyaedt/application/Design.py#L3317

Samuelopez-ansys, 2024-04-17:
@mon-kev It looks a problem with the GRPC interface. Please check this link to review different cases you could have:
https://aedt.docs.pyansys.com/version/stable/Getting_started/Troubleshooting.html

I can not reproduce the behavior, so it is difficult to fix it. Maybe there is some connection delay, because when you create the design, you are creating the PyAEDT Application object, in the last PyAEDT version we improved the load efficiency (lazy load), so maybe you could see some improvements.

closed 2024-07-26 by Samuelopez-ansys