ansys/pyaedt #4382: export_field_file() fails when default variation_dict is used, and some design variables have "sweep" disabled

nathmay, 2024-03-19:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

`export_field_file()` in pyaedt/modules/PostProcessor.py has an optional argument `variation_dict`. By default, the `export_field_file()` assigns this as: `variation_dict = self._app.available_variations.nominal_w_values_dict` (unless user passes a custom dictionary)

This works when all the design variables in the project are enabled for sweep. However, if 'sweep' is disabled for one or more design variables, `variation_dict ` should exclude those variables. Otherwise, ` ofieldsreporter.ExportToFile()` throws as error. The issue is that `app.available_variations.nominal_w_values_dict` by default includes all design variables, even those with sweep disabled. To curcumvent this, I had to manually pass a custom `variation_dict` to `export_field_file()`, including only sweep enabled variables.

PS: I have not tested this for` export_field_file_on_grid()` - which is a similar function, and has a similarly defined `variation_dict`. So it could also have the same problem.

### Steps To Reproduce

1. Extract the attached zip file
2. Open the archived aedtz project in AnsysEDT (I used version 2023R2)
3. Make sure `datapoints.pts` is in working directory, and run the following:

```
from pyaedt import Hfss
import os
app = Hfss()
options = ['IncludePtInOutput:=', False, 'RefCSName:=', 'CubeCS']
# The following will throw an error:
app.post.export_field_file("Volume_Loss_Density", filename=os.path.join(os.getcwd(),"loss.fld"), 
                           sample_points_file=os.path.join(os.getcwd(),"data_points.pts"), export_with_sample_points=options, 
                           intrinsics="2.45GHz")
```
The work around I found is to pass a `variation_dict` that includes only the sweep enabled variables. In this example, there are 12 design variables -  but only one is sweep enabled: "cube_theta", set to "45deg" as nominal. So the following works:
```
app.post.export_field_file("Volume_Loss_Density", filename=os.path.join(os.getcwd(),"loss.fld"), 
                           sample_points_file=os.path.join(os.getcwd(),"data_points.pts"), export_with_sample_points=options, 
                           intrinsics="2.45GHz", variation_dict={'cube_theta': '45deg'})
```

[Export_Field_Test.zip](https://github.com/ansys/pyaedt/files/14654718/Export_Field_Test.zip)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

```
ansys-api-fluent==0.3.22
ansys-api-platform-instancemanagement==1.0.0
ansys-fluent-core==0.19.2
ansys-platform-instancemanagement==1.1.2
ansys-pythonnet==3.1.0rc3
anyio==4.2.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
Babel==2.14.0
beartype==0.17.0
beautifulsoup4==4.12.3
bleach==6.1.0
certifi==2023.11.17
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.1
contourpy==1.2.0
cycler==0.12.1
debugpy==1.8.0
decorator==5.1.1
defusedxml==0.7.1
docker==7.0.0
et-xmlfile==1.1.0
exceptiongroup==1.2.0
executing==2.0.1
fastjsonschema==2.19.1
fiona==1.9.5
fonttools==4.47.2
fpdf2==2.7.7
fqdn==1.5.1
geopandas==0.14.2
grpcio==1.60.1
grpcio-health-checking==1.48.2
idna==3.6
imageio==2.33.1
importlib-metadata==7.0.1
ipycanvas==0.13.1
ipyevents==2.0.2
ipykernel==6.29.0
ipython==8.20.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.8.1
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.3
json5==0.9.14
jsonpointer==2.4
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
jupyter-events==0.9.0
jupyter-lsp==2.2.2
jupyter_client==8.6.0
jupyter_core==5.7.1
jupyter_server==2.12.5
jupyter_server_terminals==0.5.2
jupyterlab==4.0.11
jupyterlab-widgets==1.1.7
jupyterlab_pygments==0.3.0
jupyterlab_server==2.25.2
kiwisolver==1.4.5
lxml==5.1.0
MarkupSafe==2.1.4
matplotlib==3.8.0
matplotlib-inline==0.1.6
mistune==3.0.2
msgpack-python==0.5.6
nbclient==0.9.0
nbconvert==7.14.2
nbformat==5.9.2
nest-asyncio==1.6.0
networkx==3.2.1
notebook==7.0.7
notebook_shim==0.2.3
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.8.1
overrides==7.7.0
packaging==23.2
pandas==2.1.1
pandocfilters==1.5.1
parso==0.8.3
pillow==10.2.0
platformdirs==4.1.0
plumbum==1.8.2
pooch==1.8.0
prometheus-client==0.19.0
prompt-toolkit==3.0.43
protobuf==3.20.3
psutil==5.9.8
pure-eval==0.2.2
pyaedt==0.8.2
pycparser==2.21
pyedb==0.5.2
pyfluent==0.2.1
Pygments==2.17.2
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
python-json-logger==2.0.7
pytomlpp==1.0.13
pytz==2023.4
pyvista==0.42.2
pywin32==306
pywinpty==2.0.12
PyYAML==6.0.1
pyzmq==25.1.2
referencing==0.33.0
requests==2.31.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rpds-py==0.17.1
rpyc==6.0.0
scikit-rf==0.31.0
scipy==1.12.0
scooby==0.9.2
Send2Trash==1.8.2
shapely==2.0.2
six==1.16.0
sniffio==1.3.0
soupsieve==2.5
SRTM.py==0.3.7
stack-data==0.6.3
terminado==0.18.0
tinycss2==1.2.1
tomli==2.0.1
tornado==6.4
traitlets==5.14.1
types-python-dateutil==2.8.19.20240106
typing_extensions==4.9.0
tzdata==2023.4
uri-template==1.3.0
urllib3==2.1.0
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.13
webcolors==1.13
webencodings==0.5.1
websocket-client==1.7.0
widgetsnbextension==3.6.6
zipp==3.17.0
```

Samuelopez-ansys, 2024-03-21:
@nathmay This is expected. If the nominal is not solved, the method fails because it is not solved this variation.

Check which variation  is the nominal before calling this method:
app.available_variations.nominal_w_values_dict

This is the value that the method is using, if it is not {'cube_theta': '45deg'}, then you are passing a wrong value.

Thanks for the feedback

nathmay, 2024-03-21:
Hi Samuel, thanks for getting back! The nominal was in fact solved. This is what I meant by some variables having sweep disabled (sorry I didn't make that clear):
[image]

In this case, after the nominal configuration shown above is solved, I get the following from `app.available_variations.nominal_w_values_dict`:
```
{'cavity_width': '26.7cm',
 'cavity_depth': '27cm',
 'cavity_height': '18.8cm',
 'wg_width': '5cm',
 'wg_depth': '7.8cm',
 'wg_height': '1.8cm',
 'plate_rad': '12cm',
 'plate_t': '0.6cm',
 'plate_buffer': '1.5cm',
 'cube_dim': '4.8cm',
 'cube_theta': '45deg'}
```
The above causes export_field_file() to throw an error.
If I just pass `variation_dict = {'cube_theta': '45deg'} ` (i.e. exclude the sweep disabled variables), then the exporting works as expected.

Hope this helps!

closed 2024-10-04 by gmalinve

gmalinve, 2024-10-04:
Hi @nathmay,

Thanks for addressing this issue.
This has now been resolved in #5230 and available in main.

Hope this helps,
Kind regards,

Giulia