ansys/pyaedt #4415: A bug about python code files to exe files.

Acaptian, 2024-03-26:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I can run my code normally in the Python compiler，but there was an error when packaging it as exe.
No such file or directory: "C:Users\\AppData\\Loca\\Temp\\LMEI131242\\pyaedt\\misc\\config.schema.json"

### Steps To Reproduce

import os
import sys
from tkinter import filedialog
from PyQt5.QtWidgets import QApplication,QMainWindow,QFileDialog,QMessageBox,QProgressBar
from qt_material import apply_stylesheet,list_themes
from ui import Ui_mainWindow
import pythonnet
import pyaedt
from pyaedt import Edb
from pyaedt import Hfss3dLayout

brd_path = filedialog.askopenfilename()
brd_directory, brd_extension = os.path.splitext(brd_path)
aedb_path = brd_directory + ".aedb"

edb = Edb(edbpath=brd_path)
edb.stackup.load(file_path=xml_path)
edb.save_edb_as(aedb_path)

edb = Edb(edbpath=aedb_path)
signal_list = []
for net in edb.nets.netlist:
    if net in user_signal_net:
        signal_list.append(net)

power_list = ["GND"]
edb.cutout(signal_list=signal_list, reference_list=power_list, extent_type="Conforming", expansion_size=0.002,
           output_aedb_path=None)

edb.save_edb()
edb.close_edb()

h3d = Hfss3dLayout(projectname=aedb_path, non_graphical=False, new_desktop_session=True)

for comp_IC in user_signal_IC:
    h3d.create_ports_on_component_by_nets(component_name=comp_IC, nets=user_signal_net)

h3d.delete_setup(setupname="setup1")

setup = h3d.create_setup(setupname="setup1")

h3d.create_linear_step_sweep(setupname=setup.name, sweepname="sweep1", unit=unit, freqstart=freqstart, freqstop=freqstop,
                             step_size=step_size,save_fields=True, save_rad_fields_only=False,
                             sweep_type="Interpolating")                           
h3d.set_meshing_settings(mesh_method="Phi", enable_intersections_check=True, use_alternative_fallback=True)

h3d.save_project()
h3d.release_desktop()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

altgraph                  0.17.4   
ansys-pythonnet           3.1.0rc3 
attrs                     23.2.0   
certifi                   2024.2.2
cffi                      1.16.0
charset-normalizer        3.3.2
click                     8.1.7
click-plugins             1.1.1
cligj                     0.7.2
clr-loader                0.2.6
colorama                  0.4.6
contourpy                 1.2.0
control                   0.9.4
cycler                    0.12.1
defusedxml                0.7.1
et-xmlfile                1.1.0
fiona                     1.9.6
fonttools                 4.45.0
fpdf2                     2.7.7
geopandas                 0.14.3
idna                      3.6
imageio                   2.34.0
Jinja2                    3.1.3
jsonschema                4.21.1
jsonschema-specifications 2023.12.1
kiwisolver                1.4.5
MarkupSafe                2.1.5
matplotlib                3.8.2
networkx                  3.2.1
numpy                     1.26.2
openpyxl                  3.1.2
osmnx                     1.9.1
packaging                 23.2
pandas                    2.1.3
pefile                    2023.2.7
Pillow                    10.1.0
pip                       24.0
platformdirs              4.2.0
plumbum                   1.8.2
pooch                     1.8.1
psutil                    5.9.6
pyaedt                    0.8.3
pycparser                 2.21
pyedb                     0.6.0
pyinstaller               6.3.0
pyinstaller-hooks-contrib 2023.10
pyparsing                 3.1.1
pyproj                    3.6.1
PyQt5                     5.15.10
PyQt5-Qt5                 5.15.2
PyQt5-sip                 12.13.0
PyQt5-stubs               5.15.6.0
python-dateutil           2.8.2
pythonnet                 3.0.3
pytomlpp                  1.0.13
pytz                      2023.3
pyvista                   0.43.4
pywin32                   306
pywin32-ctypes            0.2.2
qt-material               2.14
referencing               0.33.0
requests                  2.31.0
rpds-py                   0.17.1
rpyc                      6.0.0
scikit-rf                 0.32.0
scipy                     1.11.3
scooby                    0.9.2
setuptools                68.2.2
shapely                   2.0.3
six                       1.16.0
SRTM.py                   0.3.7
tzdata                    2023.3
urllib3                   2.2.1
utm                       0.7.0
vtk                       9.2.6
wheel                     0.36.2

MaxJPRey, 2024-03-26:
@maxcapodi78 Could #4411 fix this issue?

SMoraisAnsys, 2024-03-27:
@MaxJPRey What do you mean by packaging it as exe file ? Could you describe how to do that ? Do you mean using a shebang and executing the python file as a script ?

nathmay, 2024-03-27:
@Acaptian Are you using pyinstaller? I faced a similar error in linux when trying to use pyinstaller to create an executable from a python script containing pyaedt.

Acaptian, 2024-03-28:
[nathmay](https://github.com/nathmay)
YES. I convert my python script into a standalone executable by pyinstaller, and this issue indeed just occur at the python script aontaining pyaedt. The edb part is normal .

nathmay, 2024-03-28:
@Acaptian In that case, you may be able to solve it by specifying additional hooks. [See this example](https://gist.github.com/GaryLee/d191f239acf2f47bec93).

In your case, you'd create a file named `hook-pyaedt.py` in the same directory where you'd be running pyinstaller. The contents of this file should be:

```python
from PyInstaller.utils.hooks import collect_data_files

# Instruct pyinstaller to collect data files from pyaedt package.
    datas = collect_data_files('pyaedt')
```

You'd then run pyinstaller as:
`pyinstaller --onefile --additional-hooks-dir=. yourscript.py`
where `yourscript.py` your original python script.

This worked for me for python 3.10.14, pyinstaller 6.5.0, pyaedt 0.8.2