ansys/pyaedt #4426: set the height of the cylinder

asuperwang, 2024-03-28:
When I set the height of the cylinder, I can't use the variables I set before, I can only use float.

hfss["H"] = "1.6mm"
Cylinder = hfss.modeler.create_cylinder(cs_axis="Z", position=["L1", "0", "0"], radius=0.6, height="H", numSides=0, name="Feed", matname="pec")
height[float]
Height of the cylinder.

gmalinve, 2024-03-29:
Hi @asuperwang,

What do you mean exactly?
I tested your code snippet and I was able to get a cylinder with the expected height:
[image]

Samuelopez-ansys, 2024-04-01:
@asuperwang I think you just need to modify the parameter value:

```
hfss["H"] = "1.6mm"
Cylinder = hfss.modeler.create_cylinder(cs_axis="Z", position=["L1", "0", "0"], radius=0.6, height="H", numSides=0, name="Feed", matname="pec")

hfss["H"] = "5mm"
```

Please let us know if you need anything else, or we will close the Issue

asuperwang, 2024-04-01:
> @asuperwang我认为你只需要修改参数值：
> 
> ```
> hfss["H"] = "1.6mm"
> Cylinder = hfss.modeler.create_cylinder(cs_axis="Z", position=["L1", "0", "0"], radius=0.6, height="H", numSides=0, name="Feed", matname="pec")
> 
> hfss["H"] = "5mm"
> ```
> 
> 如果您需要其他任何信息，请告诉我们，否则我们将关闭该问题

Thank you for answering my question,even though the editor shows"Expected type 'float' but got 'str'"I can still create the required model.