ansys/pyaedt #4438: set lumped_port

asuperwang, 2024-04-01:


gkorompi, 2024-04-01:
Hello @asuperwang, please try create_port_sheet=True, as the defined cicle is indeed a sheet.

gkorompi, 2024-04-09:
Hello @asuperwang, did you have time to review my suggestion??

asuperwang, 2024-04-09:
> Hello @asuperwang, did you have time to review my suggestion??

Yes, I tried it. But the lumped_port is still not created. Is my integration line not set properly?Or are some parameters set incorrectly?
[image]
[image]
[image]

gkorompi, 2024-04-10:
Hello @asuperwang the correct expression is: 
hfss.lumped_port(signal="Port",create_port_sheet=False,integration_line=[[7.6,0,0],[0.9,0,0]],impedance = 50, name="1")

The integration_line should be a list of two points as expressed above. Also, when using two point integration line, you should use create_port_sheet = False, because create_port_sheet = True (wrongly recommended before) works only for int type integration_line (i.e. = AxisDir.ZPos etc).
Lastly, make sure that both endpoints lie on the port. If this still gives you an error please share the code and design if possible.

asuperwang, 2024-04-10:
> Hello @asuperwang the correct expression is: hfss.lumped_port(signal="Port",create_port_sheet=False,integration_line=[[7.6,0,0],[0.9,0,0]],impedance = 50, name="1")
>
> The integration_line should be a [...]

After modification, lumped_port is still not created correctly.

My process of setting lumped_port is: the starting coordinates of the integration line (7.6, 0, 0), and then enter the relative coordinates (0.9, 0, 0) in the DX, DY, DZ of the status bar, but I can't get it right in the python script place setting.
I uploaded all my code in myrepository.The code is not completely finished yet. The problem lies in setting lumped_port.

https://github.com/asuperwang/pyaedt_antenna/blob/1af109287b260b42a0b9487213ae91f9468baab0/antenna.py

gkorompi, 2024-04-11:
Hello @asuperwang the port you are trying to define is outside of the object selected:
[image]

When coordinates are set correctly (inside the circle), the port is created: 
hfss.lumped_port("Port",  create_port_sheet=False, port_on_plane=True, integration_line=[[7.2, -1.4, 0], [6.8, 1.4, 0]], impedance=50, name="1")

[image]

asuperwang, 2024-04-11:
> Hello @asuperwang the port you are trying to define is outside of the object selected: [...]

I have solved this problem, thank you very much for helping me。

closed 2024-04-11 by asuperwang