ansys/pyaedt #4453: How to select all faces of an assembly except those with z=0.

jmaravin, 2024-04-03:
### Description of the feature

Hi, I am trying to select all the faces of an assembly (which is formed by several bodies) except those in the axial symmetry plane. I can select all the faces of these objects but I cannot deselect the faces which are on z=0. I can do it manully but they are about 100 faces to deselect so it takes too much time to deselect all of them. 
The code I use to select all the faces is the following: 

faces = m3d.modeler.select_allfaces_fromobjects(objects)

fwlfHS, 2024-04-09:
if all bodies have a face on the symmetry plane, you could select that face by `obj.bottom_face_z.id` put all those faces in a list `sym_face_ids` and exclude the list of symmetry faces `other_faces = [face.id for face in faces if face.id not in sym_face_ids]`

Samuelopez-ansys, 2024-04-16:
@jmaravin Does it answer your question? Please close the issue if it does.

jmaravin, 2024-04-18:
Hi @Samuelopez-ansys and @fwlfHS it does but not completely. I understand the method to implement but the code provided does not work.. I am unable to use the code "obj.bottom_face_z.id"  
Could you please help?
Jaime

Samuelopez-ansys, 2024-04-18:
@jmaravin Without a project to reproduce the workflow is quite difficult. Please share it and show what exactly you need.

jmaravin, 2024-04-19:
Hi @Samuelopez-ansys, you are right. Please see the code below. What I am trying to do is to remove the faces at z=0 of all the cylinders through the code that @fwlfHS gave me but I am unable to do so. Could you please help?

Kind regards, 
Jaime

from pyaedt import Maxwell3d
from pyaedt.modeler.cad.object3d import Object3d

Project_Name = "example"
Design_Name = "example"
Solver = "Transient"
DesktopVersion = "2023.1"
non_graphical = False

m3d = Maxwell3d(
    projectname=Project_Name,
    designname=Design_Name,
    solution_type=Solver,
    specified_version=DesktopVersion, 
    non_graphical=non_graphical,
    new_desktop_session=False)

m3d.modeler.model_units = "mm"
modeler = m3d.modeler
Plot = m3d.odesign.GetModule("ReportSetup")
m3d.modeler.set_working_coordinate_system("Global")

XPos    = 10
Zpos    = 10
Radius  = 3

for i in range(0,15):
    m3d.modeler.create_cylinder(cs_axis="Z", position=[i*XPos, 0, -Zpos], 
                                    radius=Radius, height=Zpos, name="Cylinder"+str(i+1), 
                                    matname="Copper")
    
Cylinders = m3d.modeler.get_objects_w_string("Cylinder", case_sensitive=True)
faces = m3d.modeler.select_allfaces_fromobjects(Cylinders)
m3d.assign_insulating(faces, insulation_name="Insulation")
    
m3d.release_desktop(True,True)

fwlfHS, 2024-04-19:
```python
sym_face_ids = []
for Cylinder in Cylinders:
     obj = m3d.modeler.get_object_from_name(Cylinder)
     sym_face_ids.append(obj.bottom_face_z.id)
other_faces = [face for face in faces if face not in sym_face_ids]
```

jmaravin, 2024-04-25:
Hi @fwlfHS, it works now. Thanks very much for your help. Appreciate it. 
Regards, 
Jaime

closed 2024-04-25 by jmaravin