ansys/pyaedt #4465: Bug located in hfss3dlayout unittest

ring630, 2024-04-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Got below error when running test_01_3dlayout_edb.py

[image]

### Steps To Reproduce

run test_01_3dlayout_edb.py in IDE

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

[.test]

ring630, 2024-04-05:
@Samuelopez-ansys  @SMoraisAnsys

SMoraisAnsys, 2024-04-05:
@ring630 Could you give us more information on the failing test ? As far as I know, the CI is passing.
Did you tested it with pyaedt main branch ?

ring630, 2024-04-05:
@SMoraisAnsys , it looks like it CI is passing. It is a warining in RED. You will see this warning when you run the test on your machine.

Samuelopez-ansys, 2024-04-05:
@ring630 I think this is just a warning of the coverage package. When you run the unit test locally, you are not adding the coverage parameter like in the CI, so I am not sure this is a PyAEDT issue

SMoraisAnsys, 2024-04-05:
@ring630 I'm not sure if it's an error related to pyaedt. Can you tell me how you trigger your tests ?
Do you call pytest with some specific options ?

ring630, 2024-04-05:
@Samuelopez-ansys @SMoraisAnsys , I do not have any specific options. Maybe this warning is always there. I just didn't notice. If it not something I should worry about. I wil close this issue.

closed 2024-04-12 by Samuelopez-ansys