ansys/pyaedt #4501: Submit_job to Linux Cluster

kinielsen, 2024-04-15:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Dear Team,

We are trying use the icepak.submit_job API to Submit a job from a Windows machine to a linux cluster. 
When we do it manually from Electronics Desktop we have no problems. 
However, when using the python command, AEDT does not seem to be able to find the execution path on the Linux Cluster. 
We have tried with almost any combination of path expressions (forward slash / backslash) and importing the .areg settings file with all the setting that work when submitting manually - but none seem to work. 
[image]

I seems like the API cannot account for the Linux / Windows shared path - but I might be wrong?
Do you have any tips or tricks that might help? 

Best regards, 
Kim 
   

### Steps To Reproduce

ipk = pyaedt.Icepak(project_path)
ipk.submit_job(clustername, linux_path, setting_file=settings_file)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-api-sherlock==0.1.24
ansys-pythonnet==3.1.0rc3
ansys-sherlock-core==0.5.0
attrs==23.2.0
bcrypt==4.1.2
Brotli==1.1.0
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.2.0
cryptography==42.0.5
cycler==0.12.1
dash==2.15.0
dash-core-components==2.0.0
dash-design-kit==1.6.7
dash-html-components==2.0.0
dash-snapshots==0.0.1
dash-table==5.0.0
decorator==5.1.1
defusedxml==0.7.1
Deprecated==1.2.14
dill==0.3.8
diskcache==5.6.3
et-xmlfile==1.1.0
fabric==3.2.2
fiona==1.9.5
Flask==2.2.5
Flask-Compress==1.14
fonttools==4.48.1
fpdf2==2.7.7
geopandas==0.14.3
grpcio==1.62.1
gunicorn==21.2.0
idna==3.6
imageio==2.34.0
importlib-metadata==7.0.1
invoke==2.2.0
itsdangerous==2.1.2
Jinja2==3.1.3
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
MarkupSafe==2.1.5
matplotlib==3.8.0
multiprocess==0.70.16
nest-asyncio==1.6.0
networkx==3.2.1
numpy==1.26.0
openpyxl==3.1.2
osmnx==1.9.1
packaging==23.2
pandas==2.1.1
paramiko==3.4.0
pillow==10.2.0
platformdirs==4.2.0
plotly==5.18.0
plumbum==1.8.2
pooch==1.8.0
protobuf==3.20.3
psutil==5.9.8
pyaedt==0.8.7
pycparser==2.21
pyedb==0.7.0
PyNaCl==1.5.0
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.42.2
pywin32==306
PyYAML==6.0.1
referencing==0.33.0
requests==2.31.0
retrying==1.3.4
rpds-py==0.17.1
rpyc==6.0.0
scikit-rf==0.31.0
scipy==1.12.0
scooby==0.9.2
shapely==2.0.3
six==1.16.0
SRTM.py==0.3.7
tenacity==8.2.3
typing_extensions==4.9.0
tzdata==2023.4
urllib3==2.2.0
utm==0.7.0
vtk==9.2.6
Werkzeug==3.0.1
WMI==1.5.1
wrapt==1.16.0
zipp==3.17.0

kinielsen, 2024-04-26:
Hi @Samuelopez-ansys and @dcrawforAtAnsys ,

Thanks for the reply and I am sorry for the late response. 

I tried to record script in the UI, but the script recorder does not record any actions related to "Submit Job". It simply creates an empty .py script. 
I was not able to find the log file, but the error happens around line 1916 and it throws the error "PyAEDT ERROR: AEDT shared path does not exist. Provide a full path" 

I have looked around in the code - and I do not see the pyaedt code uses the information about shared directories between Windows and Linux. Please see below. Maybe thats the problem? 
[image]

I would be happy to share the .areg file - but I am not sure that I am allowed to do so on github, due to internal security. If you have an ANSYS email - I can send it to that as we have NDA with ANSYS. 
You are also welcome to reach out to me at kinielsen@grundfos.com :-)

dcrawforAtAnsys, 2024-05-22:
@kinielsen : Could it be that the path you send as an argument to ``desktop.submit_job()`` cannot be accessed from the Linux machine?  That argument should be a network path and be accessible from the Linux environment.

dcrawforAtAnsys, 2024-05-28:
Improved error logging in #4719

closed 2024-05-28 by dcrawforAtAnsys