ansys/pyaedt #4524: AEDT doc landing page - Remove PyEDB card?

PipKat, 2024-04-17:
Given that the PyEDB API Reference section is no longer part of the PyAEDT documentation should the doc landing page be edited to:

- Rename the card with the title "AEDT API reference" to "API reference" as in the doc for all other PyAnsys libraries.
- Remove the card with the title "EDB API reference" given separate PyEDB documentation now exists. (If desired, the renamed "API reference" section could add a note with a link to the "API reference" in the EDB documentation.)
- Relabel the cheat sheets for each of these libraries according and then display them in the left navigation pane of their respective doc landing pages. (Or, would we want to wait until these cheat sheets are redo using Markdown source files and the [PyAnsys Quarto Cheat Sheet](https://github.com/ansys-internal/pyansys-quarto-cheatsheet) extension?

@MaxJPRey and @maxcapodi78 If you agree with these recommended changes, you can assign me this issue and I'll do the work.

MaxJPRey, 2024-04-18:
@PipKat I fully agree with your 3 suggestions.

Maybe at some point, we could add a link in the PyAEDT doc toward the PyEDB doc because they are related.

Adding @SMoraisAnsys  for visibility.

SMoraisAnsys, 2024-04-18:
Great suggestions ! If you want to do it, I would gladly let you handle items 1 and 2.
For the last one, we might try to work with pyansys quarto but it could be done in a second group of changes / PR.

closed 2024-06-03 by Samuelopez-ansys