ansys/pyaedt #4532: Eliminate the negative for the keyword argument ``non_graphical``

dcrawforAtAnsys, 2024-04-18:
### Description of the current limitations

``non_graphical`` unnecessarily introduces negation to the keyword. Readability would be improved if this were changed to ``graphical``. The current syntax:
- ``non_graphical = True`` -> ``graphical = False``
- ``non_graphical = False`` -> ``graphical = True``

### Code sample expected

For example:
``` python
@pyaedt_function_handler(graphical="not non_graphical")
def Circuit(
    projectname=None,
    designname=None,
    solution_type=None,
    setup_name=None,
    specified_version=None,
    graphical=True,          # <-- Change this line
    new_desktop_session=False,
    close_on_exit=False,
    student_version=False,
    machine="",
    port=0,
    aedt_process_id=None,
):
```

gmalinve, 2024-04-21:
@dcrawforAtAnsys linked [v1.0](https://github.com/ansys/pyaedt/milestone/3) and #4420

gmalinve, 2024-10-01:
@Devin-Crawford since we decided to keep it as is for now and since we closed #4420, I'll close this one too.

closed 2024-10-01 by gmalinve