ansys/pyaedt #4614: Add modeler.export_gds()

Switcherag, 2024-04-30:
### Description of the feature

Similar to import_gds() and oEditor.ExportGDSII (IronPython)

Would be useful for me.
Thanks,
Switcherag

Switcherag, 2024-06-10:
@SMoraisAnsys Can you help with this ( I saw your wanted to help inscreasing current coverage )

Samuelopez-ansys, 2024-06-10:
@Switcherag If you know how to implement this feature, you can add it and create a pull request. 

https://aedt.docs.pyansys.com/version/stable/Getting_started/Contributing.html

SMoraisAnsys, 2024-06-10:
> @SMoraisAnsys Can you help with this ( I saw your wanted to help inscreasing current coverage )

I was speaking about test coverage, which was handled by @Samuelopez-ansys in #4736
@Switcherag Ftm I don't have the time to implement new features. I'd advise you to follow Samuel's advice and implement it yourself if that's not too hard (which seems to be the case from your issue's description).
If you try it out, feel free to contact us to help you make your first contribution in `pyaedt` !

gmalinve, 2025-04-16:
Closing as per duplicate of #5827.

closed 2025-04-16 by gmalinve