ansys/pyaedt #4619: Support for "Calculator" function from Field Overlays in Icepak

BenjaminFriess, 2024-05-02:
### Description of the feature

Hello,

I would like to evaluate an expression with the calculator function from Icepak.

1) I couldn't find any function to use the calculator in Pyaedt. Would it be possible to implement the calculator in Pyaedt API, to make calculations and read the result? See example below:
[image]

2) Meanwhile, I tried to use the AEDT API with functions I found through macro recording. I managed to make the calculation with following code:

oDesign = icepak.odesign
oModule = oDesign.GetModule("FieldsReporter")
oModule.CopyNamedExprToStack("HeatFlowRate")
oModule.EnterSurf("Facelist1")
oModule.CalcOp("Integrate")
oModule.ClcEval("Setup1 : SteadyState", [], "Fields")

However, I didn't find how to read the result. Does someone know a way on how to read back the result from the calculator?

Thanks
Benjamin

BenjaminFriess, 2024-05-02:
Hi @lorenzovecchietti

You are right, the field summary would be easier to use. Unfortunately, with the field summary, when trying to get the heat flow rate through surfaces, for some unknown reason, the field summary sometimes gives wrong results. I had submitted a ticked on this topic on ANSYS customer portal, but the issue was not solved.

The calculator does not have the same issue, that's why I need to use it.

BenjaminFriess, 2024-05-02:
Hello,

Meanwhile, I managed to find a solution that is not very smart, but that works. It is inspired from PostProcessor.py, starting at line 2546. The solution is to write the result in a file, and to read back the file to get the result. See code below:

oModule.ClcEval("Setup1 : SteadyState", [], "Fields")
file_name = os.path.join(icepak.temp_directory, generate_unique_name("temp_fld") + ".fld")
oModule.CalculatorWrite(file_name, ["Solution:=", "Setup1 : SteadyState"], [])
value = None
if os.path.exists(file_name) or settings.remote_rpc_session:
    with open_file(file_name, "r") as f:
        lines = f.readlines()
        lines = [line.strip() for line in lines]
        value = lines[-1]
    os.remove(file_name)
return float(value)

It would be nice if we could get from Ansys a function that enables getting the result from the calculator, without having to write in in a file.

ansys-satyajeet, 2024-05-03:
Hi Benjamin,

You can try the following to compute the values from the fields calculator and save it to a report and then export it out. I agree that it may not be the best solution, but worth a try.

```
calc_expr = []
app.ofieldsreporter.CalcStack("clear")
for item in list_of_faces:
      app.ofieldsreporter.EnterQty("Heat_Flow_Rate")
      app.ofieldsreporter.EnterSurf(item)
      app.ofieldsreporter.CalcOp("Integrate")
      named_expr = item + "_heat_flow_rate"
      if app.ofieldsreporter.DoesNamedExpressionExists(named_expr):
          app.ofieldsreporter.DeleteNamedExpr(named_expr)
      app.ofieldsreporter.AddNamedExpression(named_expr, "Fields")
      calc_expr.append(named_expr)
  a = ["X:=", ["All"]]
  b = ["X Component:=", "X", "Y Component:=", calc_expr]
  existing_reports = app.odesign.GetChildObject("Results").GetChildNames()
  for rep in existing_reports:
      if report_name.upper() == rep.upper():
          app.report.DeleteReports(report_name)
  app.oreportsetup.CreateReport(report_name, "Fields", "Data Table", sol_name, [], a, b)
```

Here the code creates a named expression for heat flow rate calculation out of a face. It then creates a report to print out the results there. If there are multiple faces where you need the heat flow rate, you could print them out to the report table and export the data table.

Samuelopez-ansys, 2024-05-07:
@BenjaminFriess I think you want to use get_scalar_field_value method. 

Here you will find an example of how to use it:
https://aedt.docs.pyansys.com/version/stable/examples/03-Maxwell/Maxwell_Magnet.html#compute-mass-center

And here the method:
https://aedt.docs.pyansys.com/version/0.9/API/_autosummary/pyaedt.modules.AdvancedPostProcessing.PostProcessor.get_scalar_field_value.html#get-scalar-field-value

Please close the Issue if you think it can be closed.

BenjaminFriess, 2024-05-07:
Thanks, the function you mentioned works fine. For some reason, it is slower than using the field reporter and writing results to a file, but that works.

Cordialement / Kind regards
________________________________________

Benjamin Friess
Project Manager
________________________________________

Wurth Elektronik France SAS
Intelligent Power & Control Systems
3 Rue Daniel Gruber
67117 Ittenheim
France

Tel. +33 3 68 71 8680
Fax +33 3 88 11 - 2611
***@***.******@***.***>
www.we-online.de

http://www.we-online.com/twitter
http://www.we-online.com/youtube
http://www.we-online.de/facebook
http://www.we-online.com/linkedin
________________________________________

Würth Elektronik collects and processes your personal data provided by you for the purposes of completing orders and to offer you catalogues, samples and/or products.
The legal basis for the processing of your data is Article 6 (1) lit. (b) and f) GDPR (prior to entering into a contract or necessary for the performance of a contract, as well as for the purposes
of the legitimate interests). You can find additional notices on data protection at www.we-online.com/privacy/businesscontact<http://www.we-online.com/privacy/businesscontact>.
_____________________________________________

Wurth Elektronik France SAS – 1861 avenue Henri Schneider – CS70029 – 69881 MEYZIEU CEDEX
Wurth Elektronik France S.A.S à directoire et conseil de surveillance au capital de 2 000 000 € - Siret 955 505 094 00073 - APE
4652Z - Registre de Commerce de Lyon - N° TVA FR06 955 505 094

CAUTION: External Mail!

https://aedt.docs.pyansys.com/version/stable/examples/03-Maxwell/Maxwell_Magnet.html#compute-mass-center

https://aedt.docs.pyansys.com/version/0.9/API/_autosummary/pyaedt.modules.AdvancedPostProcessing.PostProcessor.get_scalar_field_value.html#get-scalar-field-value

closed 2024-05-07 by BenjaminFriess