ansys/pyaedt #4627: Improve PyAEDT coverate

Samuelopez-ansys, 2024-05-03:
### Description of the current limitations

Current PyAEDT coverage is 81%. PyAEDT could improve it if possible.

SMoraisAnsys, 2024-05-04:
Feel free to contact me if there are modules where I can help to increase the current coverage.

Acaptian, 2024-05-07:
Hi [SMoraisAnsys](https://github.com/SMoraisAnsys)]
There are some requests:
1.Add a function to import Pmap file and export syz results by Siwave in siwave module or edb module.
2.Fix the functon :
setup = edbapp.create_siwave_syz_setup("setup1")
setup.add_frequency_sweep(
    frequency_sweep=[
        ["linear count", "0", "1kHz", 1],
        ["log scale", "1kHz", "0.1GHz", 10],
        ["linear scale", "0.1GHz", "10GHz", "0.1GHz"],
    ]
)
3.Provide an example of exporting the report JSON file through the already created simulation report template.
Thanks.

Samuelopez-ansys, 2024-05-07:
Hi @Acaptian ,

Coverage means the unit test coverage. Your requests are new features, and in addition, they do not correspond to PyAEDT. You should ask for these new features in PyEDB, which is the focused in Layout automation:

https://github.com/ansys/pyedb

Regarding the 3rd request, here you can find a JSON report example:
https://aedt.docs.pyansys.com/version/stable/User_guide/pyaedt_file_data/reports.html

Acaptian, 2024-05-08:
Hi @[Samuelopez-ansys](https://github.com/Samuelopez-ansys) ,
Thanks for your reply . 
My purpose for the third point is to export the already created report template to a JSON file, so that it can be reused in the next simulation.
like this,
[image]

Samuelopez-ansys, 2024-05-08:
@Acaptian There is no method to create the JSON file because there are multiple options and it would be difficult to cover all report option, but I shared an example of a JSON file, you just need to modify the JSON file, it is straightforward, as you said, you just need to create the JSON once and then load the JSON like in this example:

https://aedt.docs.pyansys.com/version/stable/User_guide/pyaedt_file_data/reports.html

Samuelopez-ansys, 2024-05-08:
More examples:
https://aedt.docs.pyansys.com/version/stable/examples/07-Circuit/Reports.html#sphx-glr-examples-07-circuit-reports-py

and

https://github.com/ansys/pyaedt/tree/main/_unittest/example_models/report_json