ansys/pyaedt #4649: Error when closing EDT that is running on LSF

kruglovIFX, 2024-05-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

OS: Red Hat Enterprise 8.8

In pyaedt 0.8 and 0.9, `release_desktop()` and `close_desktop()` do not close EDT and return an error if the session was started via LSF via `settings.use_lsf_scheduler = True `.

When started locally with `settings.use_lsf_scheduler = False `, EDT closes without a problem.

### Steps To Reproduce

### 1) Launch EDT on LSF
Code:
```
from pyaedt import Desktop
from pyaedt import settings 
 
grpcsrv = ''                   
settings.use_lsf_scheduler = True   
settings.lsf_queue = "ssr" 
settings.lsf_num_cores = 4 
settings.lsf_ram = 4000  
settings.lsf_osrel = 80     
settings.lsf_ui = "ansysedt_gui"
 
desktop = Desktop()
```

Output:

> PyAEDT INFO: Initializing new Desktop session.
> PyAEDT INFO: StdOut is enabled
> PyAEDT INFO: Log on file is enabled
> PyAEDT INFO: Log on Desktop Message Manager is enabled
> PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
> PyAEDT INFO: Launching PyAEDT outside AEDT with gRPC plugin.
> PyAEDT INFO: New AEDT session is starting on gRPC port 35883
> ['bsub', '-n', '4', '-q', 'ssr', '-R', '"select[(osrel=80) && ui=ansysedt_gui] rusage[mem=4000]"', '-Is', 'ssrrun ansysedt', '-grpcsrv', '35883']
> PyAEDT INFO: XXX
> PyAEDT INFO: Ansoft.ElectronicsDesktop.2023.2 version started with process ID 3154983.
> PyAEDT INFO: pyaedt v0.9.0
> PyAEDT INFO: Python version 3.10.14 (main, May  6 2024, 19:42:50) [GCC 11.2.0]
> PyAEDT INFO: AEDT 2023.2.1 Build Date 2023-10-16 09:55:00

### 2) Close EDT
Code:
```
desktop.release_desktop()
```
Output:

> close_desktop = True, pid = 3154983, is_grpc_api = True
> PyAEDT INFO: Desktop has been released and closed.
> /opt/ansysem_eval/pyansys/py310/pyaedt/0.9/rh88/lib/python3.10/site-packages/pyaedt/desktop.py:261: UserWarning: Something went wrong closing AEDT. Exception in `_main.oDesktop.QuitApplication()`.
>   warnings.warn("Something went wrong closing AEDT. Exception in `_main.oDesktop.QuitApplication()`.")
> Out[2]: True

[image]

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

annotated-types==0.6.0
ansys-pythonnet==3.1.0rc3
asttokens==2.4.1
attrs==23.2.0
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
cloudpickle==3.0.0
clr-loader==0.2.6
comm==0.2.2
contourpy==1.2.1
cycler==0.12.1
debugpy==1.8.1
decorator==5.1.1
defusedxml==0.7.1
distro==1.9.0
dotnetcore2==3.1.23
et-xmlfile==1.1.0
exceptiongroup==1.2.1
executing==2.0.1
fiona==1.9.6
fonttools==4.51.0
fpdf2==2.7.8
geopandas==0.14.4
idna==3.7
imageio==2.34.1
ipykernel==6.29.4
ipython==8.24.0
jedi==0.19.1
jsonschema==4.22.0
jsonschema-specifications==2023.12.1
jupyter_client==8.6.1
jupyter_core==5.7.2
kiwisolver==1.4.5
matplotlib==3.8.4
matplotlib-inline==0.1.7
nest-asyncio==1.6.0
networkx==3.3
numpy==1.26.4
openpyxl==3.1.2
osmnx==1.9.3
packaging==24.0
pandas==2.2.2
parso==0.8.4
pexpect==4.9.0
pillow==10.3.0
platformdirs==4.2.1
plumbum==1.8.3
pooch==1.8.1
prompt-toolkit==3.0.43
psutil==5.9.8
ptyprocess==0.7.0
pure-eval==0.2.2
pyaedt==0.9.0
pycparser==2.22
pydantic==2.7.1
pydantic_core==2.18.2
pyedb==0.10.0
Pygments==2.18.0
pyparsing==3.1.2
pyproj==3.6.1
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.43.7
pyzmq==26.0.3
referencing==0.35.1
requests==2.31.0
rpds-py==0.18.1
rpyc==6.0.0
Rtree==1.2.0
scikit-rf==1.0.0
scipy==1.13.0
scooby==0.10.0
shapely==2.0.4
six==1.16.0
spyder-kernels==2.4.4
SRTM.py==0.3.7
stack-data==0.6.3
toml==0.10.2
tornado==6.4
traitlets==5.14.3
typing_extensions==4.11.0
tzdata==2024.1
urllib3==2.2.1
utm==0.7.0
vtk==9.2.6
wcwidth==0.2.13
wurlitzer==3.1.0

dcrawforAtAnsys, 2024-05-17:
Hi @kruglovIFX ,

Could you try closing the Desktop with the command 
``` python
desktop.odesktop.QuitApplication()
``` 

instead of 
``` python
desktop.release_desktop()
```

Samuelopez-ansys, 2024-06-02:
@dcrawforAtAnsys @kruglovIFX Any update regarding this issue?

kruglovIFX, 2024-06-03:
> @dcrawforAtAnsys @kruglovIFX Any update regarding this issue?

Hi @Samuelopez-ansys,

`desktop.odesktop.QuitApplication(`) closes EDT but produces errors and the LSF job hangs. I am in contact with @dcrawforAtAnsys  on this, we are looking into it in our environment. 

The problem with `release_desktop` seems to be in `desktop.py`, inside the` _close_aedt_application` method where it tries to do `os.kill(pid, 9)` to close the application. This also throws a misleading exception 

> "Something went wrong closing AEDT. Exception in `_main.oDesktop.QuitApplication()`."

even though `QuitApplication` is not used.

dcrawforAtAnsys, 2024-06-05:
It seems there was a timing issue related to the way the buffer was being read from ``subprocess.Popen()`` in ``desktop.py``. Logging for LSF was improved and the issue seems to be resolved with #4767

closed 2024-06-05 by dcrawforAtAnsys