ansys/pyaedt #4653: Bug located in create_lna_schematic_from_snp（）

Acaptian, 2024-05-09:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Solving error

### Steps To Reproduce

import os
import pyaedt
from pyaedt.modules import report_templates
from tkinter import filedialog

snp_path = filedialog.askopenfilename()
cir = pyaedt.Circuit(projectname=None, non_graphical=False)
cir.create_lna_schematic_from_snp(snp_path,
        start_frequency=0,
        stop_frequency=1.5,
        auto_assign_diff_pairs=False,
        separation=".",
        pattern=None,
        analyze=True,
        design_name="LNA")
cir.save_project()
cir.release_desktop()
[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

altgraph                  0.17.4
ansys-pythonnet           3.1.0rc3
attrs                     23.2.0
certifi                   2024.2.2
cffi                      1.16.0
charset-normalizer        3.3.2
click                     8.1.7
click-plugins             1.1.1
cligj                     0.7.2
clr-loader                0.2.6
colorama                  0.4.6
contourpy                 1.2.0
control                   0.9.4
cycler                    0.12.1
defusedxml                0.7.1
et-xmlfile                1.1.0
fiona                     1.9.6
fonttools                 4.45.0
fpdf2                     2.7.7
geopandas                 0.14.3
idna                      3.6
imageio                   2.34.0
Jinja2                    3.1.3
jsonschema                4.21.1
jsonschema-specifications 2023.12.1
kiwisolver                1.4.5
MarkupSafe                2.1.5
matplotlib                3.8.2
networkx                  3.2.1
numpy                     1.26.2
openpyxl                  3.1.2
osmnx                     1.9.1
packaging                 23.2
pandas                    2.1.3
pefile                    2023.2.7
Pillow                    10.1.0
pip                       24.0
platformdirs              4.2.0
plumbum                   1.8.2
pooch                     1.8.1
psutil                    5.9.6
pyaedt                    0.9.0
pycparser                 2.21
pyedb                     0.6.0
pyinstaller               6.5.0
pyinstaller-hooks-contrib 2024.3
pyparsing                 3.1.1
pyproj                    3.6.1
PyQt5                     5.15.10
PyQt5-Qt5                 5.15.2
PyQt5-sip                 12.13.0
PyQt5-stubs               5.15.6.0
python-dateutil           2.8.2
pythonnet                 3.0.3
urllib3                   2.2.1
utm                       0.7.0
vtk                       9.2.6
wheel                     0.36.2

Samuelopez-ansys, 2024-05-09:
@Acaptian please share more information about the SnP file. Did you check if it is imported correctly? Because maybe the format of the file is not compatible with Circuit, please first try to import manually the file.

if it works, please share the file.

Acaptian, 2024-05-09:
@Samuelopez-ansys
I tried to import it manually , and it was correct.
This is the file
[MIPI_1.5GBps_test.zip](https://github.com/ansys/pyaedt/files/15258464/MIPI_1.5GBps_test.zip)

Samuelopez-ansys, 2024-05-09:
Hi @Acaptian,

After checking the code, everything is OK, but I realized that AEDT UI changes internally the name of the file to avoid any special character, like in our case the "." in the "1.5". I have just renamed the sile test.s20p, and it works perfectly. 

I will implement a check in this method to avoid this issue.

PR #4654 (2024-05-10): REFACTOR: Touchstone name with "."
If model has a ".", Circuit solver crashes. Adding a "_" in this case.

Close #4653
closed 2024-05-10 by Samuelopez-ansys