ansys/pyaedt #4704: Avoid default plotting of matplotlib figures inside the methods.

dcrawforAtAnsys, 2024-05-21:
### Description of the current limitations

Rendering matplotlib figures inside some methods can cause delays and unexpected behavior. This also appears to have caused some random time-out failures in unit tests. The issue can be addressed by explicitly exposing the "show" keyword in all methods that return Matplotlib.pyplot ``Figure`` objects.

### Code sample expected

See for example ``pyaedt.generic.plot``. The method ``plot_2d_chart()`` generates a plot if the ``snapshot_path`` is not passed as an argument. The behavior in ``plot_polar_chart()`` is different. It will render the plot if the keyword ``show=True``. The behavior should be consistent cross methods and ``show`` should be exposed for all methods that return Matplotlib Figure objects.

SMoraisAnsys, 2024-05-21:
@dcrawforAtAnsys FYI I just started working on https://github.com/ansys/pyaedt/pull/4702 to ease our way of working with plotting figures. Current changes are a WIP (as I didn't looked at the impact on the documentation side) but they should ease the way of testing plot methods in our CI CD. The main idea is to use the backend `Agg` which is a non-GUI backend and would avoid us having test ending in timeout as no one closed the opened figure!

gmalinve, 2024-05-23:
@dcrawforAtAnsys I think this one is linked to the closed PR #4708. Correct?

dcrawforAtAnsys, 2024-05-23:
@gmalinve Yes. Exposing the keyword ``show`` in all methods makes it possible to disable rendering of Matplotlib figures. I'm closing this issue.

closed 2024-05-23 by dcrawforAtAnsys