ansys/pyaedt #4718: Bug located in "create_page_port(name="name", location=[], angle=90)"

giuseppe1692, 2024-05-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi everyone,
using this function I noticed that changing the angle from 90 to 180 degrees does not change anything. You can't do the flip unless you go through IronPython. Could you correct it as soon as possible? Thank you

### Steps To Reproduce

Use the method to create page port

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

altgraph==0.17.4
ansys-pythonnet==3.1.0rc3
anyio==4.3.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
auto-py-to-exe==2.43.3
Babel==2.14.0
beautifulsoup4==4.12.3
bleach==6.1.0
bottle==0.12.25
bottle-websocket==0.2.9
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.2.0
cycler==0.12.1
debugpy==1.8.1
decorator==5.1.1
defusedxml==0.7.1
Eel==0.16.0
et-xmlfile==1.1.0
exceptiongroup==1.2.0
executing==2.0.1
fastjsonschema==2.19.1
fiona==1.9.6
fonttools==4.49.0
fpdf2==2.7.8
fqdn==1.5.1
future==1.0.0
geopandas==0.14.3
gevent==24.2.1
gevent-websocket==0.10.1
greenlet==3.0.3
h11==0.14.0
httpcore==1.0.4
httpx==0.27.0
[... 86 lines cut ...]
stack-data @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/stack_data-0.6.3-py3-none-any.whl#sha256=d5558e0c25a4cb0853cddad3d77da9891a08cb85dd9f9f91b9f8cd66e511e695
terminado @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/terminado-0.18.0-py3-none-any.whl#sha256=87b0d96642d0fe5f5abd7783857b9cab167f221a39ff98e3b9619a788a3c0f2e
tinycss2 @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/tinycss2-1.2.1-py3-none-any.whl#sha256=2b80a96d41e7c3914b8cda8bc7f705a4d9c49275616e886103dd839dfc847847
tomli @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/tomli-2.0.1-py3-none-any.whl#sha256=939de3e7a6161af0c887ef91b7d41a53e7c5a1ca976325f429cb46ea9bc30ecc
tornado @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/tornado-6.4-cp38-abi3-win_amd64.whl#sha256=10aeaa8006333433da48dec9fe417877f8bcc21f48dda8d661ae79da357b2a63
traitlets @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/traitlets-5.14.1-py3-none-any.whl#sha256=2e5a030e6eff91737c643231bfcf04a65b0132078dad75e4936700b213652e74
types-python-dateutil @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/types_python_dateutil-2.8.19.20240106-py3-none-any.whl#sha256=efbbdc54590d0f16152fa103c9879c7d4a00e82078f6e2cf01769042165acaa2
typing_extensions @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/typing_extensions-4.10.0-py3-none-any.whl#sha256=69b1a937c3a517342112fb4c6df7e72fc39a38e7891a5730ed4985b5214b5475
tzdata @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/tzdata-2024.1-py2.py3-none-any.whl#sha256=9068bc196136463f5245e51efda838afa15aaeca9903f49050dfa2679db4d252
uri-template @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/uri_template-1.3.0-py3-none-any.whl#sha256=a44a133ea12d44a0c0f06d7d42a52d71282e77e2f937d8abd5655b8d56fc1363
urllib3 @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/urllib3-2.2.1-py3-none-any.whl#sha256=450b20ec296a467077128bff42b73080516e71b56ff59a60a02bef2232c4fa9d
utm @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/utm-0.7.0-py3-none-any.whl#sha256=19457f175c8d79a32d594b3c2571c8204e65cd0b8eebf7be76811c9ee1f8525b
virtualenv-clone==0.5.7
vtk @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/vtk-9.3.0-cp310-cp310-win_amd64.whl#sha256=2aae778297817335ddd6698b4c124c109d8ac476512691fe19446614ae43ba56
wcwidth @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/wcwidth-0.2.13-py2.py3-none-any.whl#sha256=3da69048e4540d84af32131829ff948f1e022c1c6bdb8d6102117aac784f6859
webcolors @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/webcolors-1.13-py3-none-any.whl#sha256=29bc7e8752c0a1bd4a1f03c14d6e6a72e93d82193738fa860cbff59d0fcc11bf
webencodings @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/webencodings-0.5.1-py2.py3-none-any.whl#sha256=a0af1213f3c2226497a97e2b3aa01a7e4bee4f403f95be16fc9acd2947514a78
websocket-client @ file:///C:/Users/giuseppe.giordano/Desktop/venv_giuseppe/venv_giuseppe/libreria_esterna/websocket_client-1.7.0-py3-none-any.whl#sha256=f4c3d22fec12a2461427a29957ff07d35098ee2d976d3ba244e688b8b4057588
whichcraft==0.6.1
xlrd==2.0.1
XlsxWriter==3.2.0
xlutils==2.0.0
xlwt==1.3.0
zope.event==5.0
zope.interface==6.2

gmalinve, 2024-05-23:
Hi @giuseppe1692,

you can always access the angle property in pyaedt after you created it: 
`my_port = self.aedtapp.modeler.schematic.create_page_port("my_port", [0,0,0], 0)`
`my_port.angle = 90`

Hope this helps,

Giulia

giuseppe1692, 2024-05-23:
Hi @gmalinve,
thank you for your answer. 
Unfortunately it doesn't work and I get this warning: 
PyAEDT WARNING: Grpc doesn't support angle settings because special characters are not supported.

gmalinve, 2024-05-23:
@giuseppe1692, ok let me investigate.
In the meantime I see you are still using an older version of pyaedt. Could you please upgrade to the latest and re-try?

giuseppe1692, 2024-05-23:
Hi @gmalinve 
I've updgraded to 0.9.1 but there is the same issue.

Below the code: 

```python
resistor = aedt_app.modeler.schematic.create_resistor(value=0, location=[offset_x + 1000, offset_y])
page_port = aedt_app.modeler.schematic.create_page_port(name="DUT_" + str(site) + "_" + lista_net_needle[i], location=[offset_x+1500,offset_y], angle=90)
page_port.angle = 180 
```

gmalinve, 2024-05-23:
@giuseppe1692 which version of AEDT are you using?

gmalinve, 2024-05-23:
@giuseppe1692 I suppose you are using version < 2023.1.

giuseppe1692, 2024-05-24:
Hi @gmalinve,
On my local PC I'm using 2023.2
I'll try 2024.1 and let you know.

gmalinve, 2024-05-24:
@giuseppe1692 I'm pretty sure with 2024.1 you won't have any issue.
The fact is that all versions previous to 2024.1 (2023.2 included), if launched in grpc mode, don't allow this parameter to be changed.

closed 2024-05-27 by Samuelopez-ansys