ansys/pyaedt #4728: Bug located in ... get_antenna_ffd_solution_data

pceato, 2024-05-27:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am trying to extract the fields of a setup with two sweeps in this way:

freq = np.linspace(fmin, fmax, numFreqs).tolist()
data = hfss.get_antenna_ffd_solution_data(frequencies=freq, 
                                          setup='Setup1 : Sweep',
                                          sphere="Infinite Sphere1")

When executing this code I get the error that the number of frequencies are different from the number of eep files. I have also tried to put the frequencies of the other sweep and change the setup, but I don't get it, what am I missing? thanks a lot.

### Steps To Reproduce

1. Create an interactive instance of HFSS and connect to an existing HFSS design
hfss = pyaedt.Hfss(projectname="prueba",
                              designname="2x2", 
                              specified_version="2023.2", 
                              non_graphical="False")
OK
2. Create the variable freq
freq = np.linspace(fmin, fmax, numFreqs).tolist()
OK
3. Export the antenna parameters to Far Field Data (FFD) files 
data = hfss.get_antenna_ffd_solution_data(frequencies=freq, 
                                                                     setup='Setup1 : Sweep',
                                                                     sphere="Infinite Sphere1")
ERROR
[image]

4. Release desktop
hfss.release_desktop()
OK

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

WARNING: Skipping C:\ProgramData\anaconda3\Lib\site-packages\vtk-9.0.3.egg-info due to invalid metadata entry 'name'
aext_assistant @ file:///C:/b/abs_05zbnpu9if/croot/aext-assistant_1698112951886/work
aext_assistant_server @ file:///C:/b/abs_83f6_dptu3/croot/aext-assistant-server_1698104652697/work/backend_lib/assistant
aext_core @ file:///C:/b/abs_abml9re1ky/croot/aext-core_1698107590809/work
aext_core_server @ file:///C:/b/abs_dev7ac1xol/croot/aext-core-server_1698098831843/work/backend_lib/core
aext_shared @ file:///C:/b/abs_306_rwdm13/croot/aext-shared_1698068353440/work/backend_lib/shared
aiobotocore @ file:///C:/b/abs_74o47svlua/croot/aiobotocore_1680004300264/work
aiofiles @ file:///C:/b/abs_9ex6mi6b56/croot/aiofiles_1683773603390/work
aiohttp @ file:///C:/ci_311/aiohttp_1676432932774/work
aioitertools @ file:///tmp/build/80754af9/aioitertools_1607109665762/work
aiosignal @ file:///tmp/build/80754af9/aiosignal_1637843061372/work
aiosqlite @ file:///C:/b/abs_9djc_0pyi3/croot/aiosqlite_1683773915844/work
alabaster @ file:///home/ktietz/src/ci/alabaster_1611921544520/work
anaconda-anon-usage @ file:///C:/b/abs_ffxflkd4bw/croot/anaconda-anon-usage_1695037769762/work
anaconda-catalogs @ file:///C:/b/abs_8btyy0o8s8/croot/anaconda-catalogs_1685727315626/work
anaconda-client==1.12.0
anaconda-cloud-auth @ file:///C:/b/abs_27x7fzodlx/croot/anaconda-cloud-auth_1712794918678/work
anaconda-navigator @ file:///C:/b/abs_d8d4a02c3t/croot/anaconda-navigator_1713464063970/work
anaconda-project @ file:///C:/ci_311/anaconda-project_1676458365912/work
anyio @ file:///C:/ci_311/anyio_1676425491996/work/dist
appdirs==1.4.4
argon2-cffi @ file:///opt/conda/conda-bld/argon2-cffi_1645000214183/work
[... 334 lines cut ...]
ujson @ file:///C:/ci_311/ujson_1676434714224/work
Unidecode @ file:///tmp/build/80754af9/unidecode_1614712377438/work
urllib3 @ file:///C:/b/abs_889_loyqv4/croot/urllib3_1686163174463/work
w3lib @ file:///Users/ktietz/demo/mc3/conda-bld/w3lib_1629359764703/work
watchdog @ file:///C:/ci_311/watchdog_1676457923624/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
websocket-client @ file:///C:/ci_311/websocket-client_1676426063281/work
Werkzeug @ file:///C:/b/abs_8578rs2ra_/croot/werkzeug_1679489759009/work
whatthepatch @ file:///C:/ci_311/whatthepatch_1678402578113/work
widgetsnbextension @ file:///C:/b/abs_882k4_4kdf/croot/widgetsnbextension_1679313880295/work
win-inet-pton @ file:///C:/ci_311/win_inet_pton_1676425458225/work
win32-setctime @ file:///home/conda/feedstock_root/build_artifacts/win32_setctime_1642883564726/work
wrapt @ file:///C:/ci_311/wrapt_1676432805090/work
xarray @ file:///C:/b/abs_5bkjiynp4e/croot/xarray_1689041498548/work
xlwings @ file:///C:/ci_311_rebuilds/xlwings_1679013429160/work
xyzservices @ file:///C:/ci_311/xyzservices_1676434829315/work
y-py @ file:///C:/b/abs_b7f5go6r0j/croot/y-py_1683662173571/work
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
yarl @ file:///C:/ci_311/yarl_1676432870023/work
ypy-websocket @ file:///C:/b/abs_4e65ywlnv8/croot/ypy-websocket_1684172103529/work
zict @ file:///C:/b/abs_fc7elavmem/croot/zict_1682698759288/work
zipp @ file:///C:/ci_311/zipp_1676426100491/work
zope.interface @ file:///C:/ci_311/zope.interface_1676439868776/work
zstandard==0.19.0

Samuelopez-ansys, 2024-05-27:
@pceato Did you check the solved frequencies? 

In your screenshot, it looks like the setup do not exist:

[image]

Please provide a dummy example, if you do not find the issue. Try to debug this example, and maybe you will find what is wrong in your design.

https://aedt.docs.pyansys.com/version/stable/examples/02-HFSS/Array.html#sphx-glr-examples-02-hfss-array-py

pceato, 2024-05-28:
Thanks for your answer @Samuelopez-ansys 

Yes, I checked the frequencies and they are the ones that were in the sweep.

I can't run that example because of incompatibility with version 2023.2.

I have been trying to export the fields with one of the ansys example files and I have been able to do it using the jupyter that opens from adt desktop as shown in the image.

[image]

 If I try to do it from a local script with another environment (spyder in my case), it gives me an error and when I do the release_desktop it deletes the project designs.
[image]
[image]

I don't understand what could be happening if the code used in both cases is the same, except for the “desktop” class generated by the jupyter.
[image]
[image]

Samuelopez-ansys, 2024-05-28:
@pceato 

If it works with the example, and with your project, it means you have some settings not correct, I guess the frequencies you are passing are not solved. If you debug your code, does data contains anything?

I played with the following examples, and it works as well:

```
import os
import pyaedt

project_name = pyaedt.generate_unique_project_name(project_name="dipole")

aedt_version = "2023.2"

non_graphical = False

d = pyaedt.launch_desktop(aedt_version, non_graphical=non_graphical, new_desktop_session=True)

hfss = pyaedt.Hfss(projectname=project_name, solution_type="Modal")

hfss["l_dipole"] = "13.5cm"

compfile = hfss.components3d["Dipole_Antenna_DM"]
geometryparams = hfss.get_components3d_vars("Dipole_Antenna_DM")
geometryparams["dipole_length"] = "l_dipole"
hfss.modeler.insert_3d_component(compfile, geometryparams)

hfss.create_open_region(frequency="1GHz")

setup = hfss.create_setup("MySetup")
setup.props["Frequency"] = "1GHz"
setup.props["MaximumPasses"] = 1
hfss.create_linear_count_sweep(setup=setup.name, units="GHz", start_frequency=0.5, stop_frequency=1.5,
                               num_of_freq_points=251, name="sweep1", save_fields=False, sweep_type="Interpolating",
                               interpolation_tol=3, interpolation_max_solutions=255)

hfss.analyze_setup("MySetup")

ffdata = hfss.get_antenna_ffd_solution_data(frequencies=["1000MHz"], setup=hfss.nominal_adaptive,
                                            sphere="Sphere_Custom")

ffdata.plot_2d_cut(quantity='RealizedGain', primary_sweep="theta", secondary_sweep_value=0, title='FarField',
                   quantity_format="dB20", is_polar=True)

ffdata.polar_plot_3d_pyvista()

d.release_desktop()
```

As you can see it works in 2023.2, so it is the same version as yours, try passing the frequencies as string like in my previous example. If does not work, please share a dummy project, to reproduce the issue.

pceato, 2024-05-29:
Hi @Samuelopez-ansys ,

Sorry if I explained myself wrong, the problem I currently have is no longer the initial one. I have managed to export the files with my project if I open Jupyter from the AEDT window. However, when I do it from Spyder, no.

With several tests that I have carried out, I believe that the problem is in the Hfss function or in the Desktop function that, whenever I try to open a file with a specific path, it creates a file with that same name (that is, it overwrites the file that I I have on that route with that name). On the other hand, when I do it from Jupyter launch with AEDT, since a session is generated from that file, I can access it. How can I solve that?

Samuelopez-ansys, 2024-05-29:
Hi @pceato ,

I think your problem is related to the project path. In python, you need to put "//" or a "r" at the beginning of any path.

https://docs.python.org/3/tutorial/introduction.html#text

For example in your case:

projectname = r"C:/Users/pcea/...."

Let me know if it works.

pceato, 2024-05-30:
Hi @Samuelopez-ansys 

I have tried with both, when I use "//" I get the attached message error, when I use "r" it generates a new error, as you can see in the second image. Any more ideas?

[image]

[image]

pceato, 2024-05-30:
The ".aedt" extension was missing from the path, now it is working. Thank you very much, @Samuelopez-ansys

closed 2024-05-30 by pceato