ansys/pyaedt #4748: Edit sources in ffd solution data

pceato, 2024-05-31:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am trying to edit sources in the ffd solution data, but I am not getting the gain values updated in the module solution. The code I used is as follows.

```
freq = np.linspace(0.9e9, 2.7e9, 3).tolist()

eep = [r'C:\Users\pcea\Desktop\R\posprocesado\pyAnsys\prueba.pyaedt\Array\Setup1_Sweep1-SphereNew-900000000.0\eep\eep.txt',
       r'C:\Users\pcea\Desktop\R\posprocesado\pyAnsys\prueba.pyaedt\Array\Setup1_Sweep1-SphereNew-1800000000.0\eep\eep.txt',
       r'C:\Users\pcea\Desktop\R\posprocesado\pyAnsys\prueba.pyaedt\Array\Setup1_Sweep1-SphereNew-2700000000.0\eep\eep.txt']

data = []
for ii in np.arange(len(freq)):
    data.append(FfdSolutionData(eep[ii], freq[ii]))

magNew= []
for ii in np.arange(len(data[0].mag_offset)):
    if ii%2 == 0:
        magNew.append(1)
    else:
        magNew.append(0)

data[0].mag_offset = magNew

data[0].polar_plot_3d_pyvista()
```

### Steps To Reproduce

1. Import the ffd solutions data
2. Edit the sources of the array
3. Plot the fields

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

alabaster==0.7.16
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
anyio==4.3.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
astroid==3.2.2
asttokens==2.4.1
async-lru==2.0.4
atomicwrites==1.4.1
attrs==23.2.0
autopep8==2.0.4
Babel==2.15.0
backports.tarfile==1.1.1
bcrypt==4.1.3
beautifulsoup4==4.12.3
binaryornot==0.4.4
black==24.4.2
bleach==6.1.0
certifi==2024.2.2
cffi==1.16.0
chardet==5.2.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
cloudpickle==3.0.0
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.2.1
cookiecutter==2.6.0
cryptography==42.0.7
cycler==0.12.1
debugpy==1.8.1
decorator==5.1.1
defusedxml==0.7.1
diff-match-patch==20230430
dill==0.3.8
docstring-to-markdown==0.15
docutils==0.21.2
et-xmlfile==1.1.0
exceptiongroup==1.2.1
executing==2.0.1
fastjsonschema==2.19.1
fiona==1.9.6
flake8==7.0.0
fonttools==4.51.0
fpdf2==2.7.9
fqdn==1.5.1
geopandas==0.14.4
h11==0.14.0
httpcore==1.0.5
httpx==0.27.0
idna==3.7
imageio==2.34.1
imagesize==1.4.1
importlib_metadata==7.1.0
inflection==0.5.1
intervaltree==3.1.0
ipycanvas==0.13.2
ipyevents==2.0.2
ipykernel==6.29.4
ipython==8.24.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.8.1
isoduration==20.11.0
isort==5.13.2
jaraco.classes==3.4.0
jaraco.context==5.3.0
jaraco.functools==4.0.1
jedi==0.19.1
jellyfish==1.0.3
Jinja2==3.1.4
json5==0.9.25
jsonpointer==2.4
jsonschema==4.22.0
jsonschema-specifications==2023.12.1
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyter_server==2.14.0
jupyter_server_terminals==0.5.3
jupyterlab==4.2.1
jupyterlab-widgets==1.1.7
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.2
keyring==25.2.1
kiwisolver==1.4.5
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.8.4
matplotlib-inline==0.1.7
mccabe==0.7.0
mdurl==0.1.2
mistune==3.0.2
more-itertools==10.2.0
mypy-extensions==1.0.0
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
nest-asyncio==1.6.0
networkx==3.3
notebook==7.2.0
notebook_shim==0.2.4
numpy==1.26.4
numpydoc==1.7.0
openpyxl==3.1.2
osmnx==1.9.3
overrides==7.7.0
packaging==24.0
pandas==2.2.2
pandocfilters==1.5.1
paramiko==3.4.0
parso==0.8.4
pathspec==0.12.1
pexpect==4.9.0
pickleshare==0.7.5
pillow==10.3.0
platformdirs==4.2.2
pluggy==1.5.0
plumbum==1.8.3
pooch==1.8.1
prometheus_client==0.20.0
prompt-toolkit==3.0.43
psutil==5.9.8
ptyprocess==0.7.0
pure-eval==0.2.2
pyaedt==0.9.1
pycodestyle==2.11.1
pycparser==2.22
pydantic==2.7.1
pydantic_core==2.18.2
pydocstyle==6.3.0
pyedb==0.11.0
pyflakes==3.2.0
Pygments==2.18.0
pylint==3.2.2
pylint-venv==3.0.3
pyls-spyder==0.4.0
PyNaCl==1.5.0
pyparsing==3.1.2
pyproj==3.6.1
PyQt5==5.15.10
PyQt5-Qt5==5.15.2
PyQt5-sip==12.13.0
PyQtWebEngine==5.15.6
PyQtWebEngine-Qt5==5.15.2
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
python-lsp-black==2.0.0
python-lsp-jsonrpc==1.1.2
python-lsp-server==1.11.0
python-slugify==8.0.4
pytomlpp==1.0.13
pytoolconfig==1.3.1
pytz==2024.1
pyvista==0.43.8
pywin32==306
pywin32-ctypes==0.2.2
pywinpty==2.0.13
PyYAML==6.0.1
pyzmq==26.0.3
QDarkStyle==3.2.3
qstylizer==0.2.3
QtAwesome==1.3.1
qtconsole==5.5.2
QtPy==2.4.1
referencing==0.35.1
requests==2.32.2
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.7.1
rope==1.13.0
rpds-py==0.18.1
rpyc==6.0.0
Rtree==1.2.0
scikit-rf==1.0.0
scipy==1.13.1
scooby==0.10.0
Send2Trash==1.8.3
shapely==2.0.4
six==1.16.0
sniffio==1.3.1
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.5
Sphinx==7.3.7
sphinxcontrib-applehelp==1.0.8
sphinxcontrib-devhelp==1.0.6
sphinxcontrib-htmlhelp==2.0.5
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.7
sphinxcontrib-serializinghtml==1.1.10
spyder==5.5.4
spyder-kernels==2.5.1
SRTM.py==0.3.7
stack-data==0.6.3
tabulate==0.9.0
terminado==0.18.1
text-unidecode==1.3
textdistance==4.6.2
three-merge==0.1.1
tinycss2==1.3.0
toml==0.10.2
tomli==2.0.1
tomlkit==0.12.5
tornado==6.4
traitlets==5.14.3
types-python-dateutil==2.9.0.20240316
typing_extensions==4.11.0
tzdata==2024.1
ujson==5.10.0
uri-template==1.3.0
urllib3==2.2.1
utm==0.7.0
vtk==9.2.6
watchdog==4.0.1
wcwidth==0.2.13
webcolors==1.13
webencodings==0.5.1
websocket-client==1.8.0
whatthepatch==1.0.5
widgetsnbextension==3.6.6
yapf==0.40.2
zipp==3.19.0

Samuelopez-ansys, 2024-05-31:
@pceato This should work, please share in a zip file the information.

pceato, 2024-05-31:
Hi, @Samuelopez-ansys :

I am attaching [example.zip](https://github.com/user-attachments/files/15512689/example.zip) with the eep, aedtz, and py files. As you can see in these graphs, with python my gain is not changing.

[image]
[image]

Thanks.

Samuelopez-ansys, 2024-05-31:
Hi @pceato ,

This is a Python problem, when you create:

 dataAux = data

You are not creating a new object, it means that if you change something later in dataAux, data will also be modfied. 

For information:
https://www.geeksforgeeks.org/copy-python-deep-copy-shallow-copy/

In your script, then dataAux[0].mag_offset and data[0].mag_offset has exactly the same information, because they are exactly the same.

What you need is a copy of the object:

dataAux = `copy.deepcopy(data)`

[image]

pceato, 2024-05-31:
@Samuelopez-ansys It works! Thank you very much!