ansys/pyaedt #4768: Submit job not working from a project build with pyaedt after release_desktop(close_projects=False, close_desktop=False)

amichel0205, 2024-06-04:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

A project is created with pyaedt it could be HFSS, HFSS 3DL, Q3D(not tested with other AEDT solver).
graphical mode is on and at the end of the script the command q.release_desktop(close_projects=False, close_desktop=False) is used such the user has the end on the project.
User then try to submit job.
A message "Unable to submit job" is sent.

It seems the windows asking if the project can be close issued when submitting a job does not pop up.

For information if you close the project and close AEDT. Launch AEDT manually, open same project and submit job it works.

### Steps To Reproduce

See above

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

(venv_310) bash-4.2$ python -m pip freeze
annotated-types==0.6.0
ansys-pythonnet==3.1.0rc3
attrs==23.2.0
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
contourpy==1.2.1
cycler==0.12.1
defusedxml==0.7.1
distro==1.9.0
dotnetcore2==3.1.23
et-xmlfile==1.1.0
fiona==1.9.6
fonttools==4.51.0
fpdf2==2.7.8
geopandas==0.14.4
idna==3.7
imageio==2.34.1
jsonschema==4.22.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
matplotlib==3.8.4
networkx==3.3
numpy==1.26.4
openpyxl==3.1.2
osmnx==1.9.3
packaging==24.0
pandas==2.2.2
pillow==10.3.0
platformdirs==4.2.1
plumbum==1.8.3
pooch==1.8.1
psutil==5.9.8
pyaedt==0.9.1
pycparser==2.22
pydantic==2.7.1
pydantic_core==2.18.2
pyedb==0.10.0
pyparsing==3.1.2
pyproj==3.6.1
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.43.7
referencing==0.35.1
requests==2.31.0
rpds-py==0.18.1
rpyc==6.0.0
Rtree==1.2.0
scikit-rf==1.0.0
scipy==1.13.0
scooby==0.10.0
shapely==2.0.4
six==1.16.0
SRTM.py==0.3.7
toml==0.10.2
typing_extensions==4.11.0
tzdata==2024.1
urllib3==2.2.1
utm==0.7.0
vtk==9.2.6
(venv_310) bash-4.2$

Samuelopez-ansys, 2024-08-12:
Hi @amichel0205, any news regarding this Issue? I think it is an AEDT issue in Linux.

amichel0205, 2025-09-08:
I will check if the issue still exists.