ansys/pyaedt #4775: Revisiting project files

jorgepiloto, 2024-06-06:
PyAEDT has been now public for more than two years. Since its public release, new tools and workflows have been discovered. Some of them have been adopted by the PyAnsys ecosystem.

Therefore, the goal of this meta-issue is to collect some issues that can be addressed within this project.

Please, open a pull-request per issue (if possible), to speed up reviewing process.

- [ ] **Verify proper metadata in pyproject.toml**
  Ensure authors and maintainers point to the `pyansys.core@ansys.com` email and that Ansys, Inc. is the main author. Update minimum Python version. Expand project classifiers. Migrate project tools configuration to this file.

- [ ] **Consider using Ruff for linting and formatting**
  [Ruff](https://docs.astral.sh/ruff/) collects all popular tools (black, isort, flake8, codespell, and more!) under a unique project. In addition, it is extremely fast as it operates on top of Rust. This would allow to remove files like `.flake8` from the root of the project.

- [ ] **Use ansys-pre-commit-hooks**
  The [Ansys pre-commit-hooks](https://github.com/ansys/pre-commit-hooks) ensures that a project is compatible with the latest PyAnsys guidelines.

- [ ] **Unify ignore words with Vale's accept.txt file**
  The `accept.txt` file from Vale can be feed into codespell as the `ignore_words.txt` file.

- [ ] **Revisit doc build**
  There is a `*.bat` file inside of the `doc/source/` dir. This file is not required. The `make.bat` and `Makefiles` install Python libraries. This should take place on the `pyproject.toml`. Both `make.bat` and `Makefile` are generated by the `sphinx-quickstart` command. However, modern tools like Tox avoid having these files. Finally, sphinx-autoapi can be used to render a class per page in the documentation, removing unnecessary RST files.

- [ ] **Dynamic doc build**
  Consider using `sphinx-jinja` to conditionally build documentation sections via environment variables. This speeds up the development of the docs by allowing developers to exclude sections like API and Examples.

- [ ] **Implement a tox.ini file**
  [Tox](https://tox.wiki/en/latest) is a tool for automating tasks. It is platform aware and executes Python code in a dedicated venv, protecting your own Python environment.

- [ ] **Dynamic CI/CD**
  By playing with labels, labeler and events on the CI/CD, it is possible to alleviate the workload of the self-hosted runner. This way, the machine only runs the right workflows, avoiding unnecessary computations.

jorgepiloto, 2024-06-06:
Note that I am more than happy to help with any of these issues. Just wanted to collect all of them under a common meta-issue. We are doing the same with other mature projects of the ecosystem.

SMoraisAnsys, 2024-06-06:
Thanks for this issue, that's a nice reminder of what we should pay attention to.
I'll tackle this (with anyone wanting to help) in the incoming days.
@jorgepiloto We need some changes in our doc though as we can't use ansys/doc-build function for two reasons:
- problem with OS limit on open files, which prevent us to use ansys actions v6 (conflict with upload/download artifact actions)
- adding assets is still not merged (https://github.com/ansys/actions/pull/467)

I'll try to allocate time next week to work on using sphinx autoapi and see if that's give us some opportunity to update our CI

jorgepiloto, 2024-06-06:
Thanks for linking those references here, Sebastien. We can explore