ansys/pyaedt #4800: `solve_in_batch`  is not reopening current project

nathmay, 2024-06-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

If I submit an analysis with `solve_in_batch=True`, the file is closed and then submitted as a batch job correctly. But it is never reopened once the solution is over. 

See the following:

```python
>>> app.analyze(solve_in_batch=True)
PyAEDT INFO: Closing the AEDT Project test_project
PyAEDT INFO: Project test_project closed correctly
PyAEDT INFO: Solving model in batch mode on localhost
PyAEDT INFO: Batch job finished.
# Console waited here for the job to get over
True
>>> app.info
{'PyAEDT Version': '0.9.3', 'Product': 'Ansys Electronics Desktop 2024.1', 'Design Type': 'HFSS', 'Solution Type': 'Modal', 'Project Name': None, 'Design Name': None, 'Project Path': '', 'Platform': 'Windows-10-10.0.19045-SP0', 'Python Version': '3.11.6', 'AEDT Process ID': 76104, 'AEDT GRPC Port': 52101}
>>> app.project_name is None
True
```
I think this could be solved by adding a line that re-opens the project at the end of `application.Analysis.solve_in_batch()` .

Also I get a log output `PyAEDT INFO: Batch job finished` immediately after the job starts, and not when it's actually over. I think this is due to the following section in `application.Analysis.solve_in_batch()`:
```python
if run_in_thread and is_windows:
           ....
        else:
            subprocess.Popen(batch_run)
            self.logger.info("Batch job finished.")
```

### Steps To Reproduce

1. Load any AEDT project in non graphical mode. (I used an HFSS project)
2. Start analysis in batch mode with `app.analyze(solve_in_batch=True)`

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

about-time==4.2.1
alive-progress==3.1.5
annotated-types==0.7.0
ansys-api-fluent==0.3.24
ansys-api-platform-instancemanagement==1.1.0
ansys-fluent-core==0.20.1
ansys-platform-instancemanagement==1.1.2
ansys-pythonnet==3.1.0rc3
ansys-units==0.3.2
attrs==23.2.0
beartype==0.17.2
certifi==2024.6.2
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
defusedxml==0.7.1
docker==6.1.3
fonttools==4.53.0
fpdf2==2.7.9
grapheme==0.6.0
grpcio==1.64.1
grpcio-health-checking==1.62.2
idna==3.7
importlib_metadata==7.1.0
jsonschema==4.22.0
jsonschema-specifications==2023.12.1
lxml==4.9.4
numpy==1.26.4
packaging==24.1
pandas==1.5.3
pillow==10.3.0
platformdirs==3.11.0
plumbum==1.8.3
protobuf==4.25.3
psutil==5.9.8
pyaedt==0.9.3
pycparser==2.22
pydantic==2.7.4
pydantic_core==2.18.4
pyedb==0.13.0
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.1
pywin32==306
PyYAML==6.0.1
referencing==0.35.1
requests==2.32.3
rpds-py==0.18.1
rpyc==6.0.0
Rtree==1.2.0
ruamel.yaml==0.18.6
ruamel.yaml.clib==0.2.8
six==1.16.0
toml==0.10.2
typing_extensions==4.12.2
urllib3==2.2.1
websocket-client==1.8.0
zipp==3.19.2

Samuelopez-ansys, 2024-06-13:
Hi @nathmay, this is expected behavior.

Solve in batch is equivalent to launch the project and solve using the Terminal:

aedt.exe -ng -BatchSolve"

You have another parameter: run_in_thread

Which allows you to wait until the job is finished or go ahead.

There is a bug in the documentation, because it says it reopens the project, but this is not expected. The user should open the project again once it finishes, because there are multiple cases (once it finish do the post processing in non graphical mode, opening the project again, do nothing...).

So we will fix the docstring of the analyze method :

solve_in_batch : bool, optional
            Whether to solve the project in batch or not.
            If ``True`` the project will be saved, closed, and solved.

nathmay, 2024-06-13:
@Samuelopez-ansys Understood, thanks for the clarification. I'll go ahead and close this issue. 

Interestingly, I didn't see any difference in behavior between `run_in_thread=True` or `False`. This was in windows, with `solve_in_batch` and on localhost. But will investigate further before getting back.

closed 2024-06-13 by nathmay

Samuelopez-ansys, 2024-06-14:
@nathmay I review the code, and the run_in_thread is just adding the flag DETACHED_PROCESS  to the subprocess command:

The DETACHED_PROCESS flag (value 0x00000008) is a Windows-specific flag that, when set, means that the new process does not inherit the console of the parent process. The new process is given its own console, or it runs without a console, which means it will not display any command window even if it requires console input.

But then the code should wait until the file .q.completed exists.

If you do not want to wait until the design is solved, you can use the submit_job method which probably is what you need.