ansys/pyaedt #4801: Make sorting optional for `DataSet.update()`

nathmay, 2024-06-12:
### Description of the feature

The `update()` method in `application.Variables.DataSet` class sorts the dataset by it's X coordinate values - both for 1D and 3D datasets. While this makes sense for 1D, could be problematic for 3D datasets. I was only updateing the `v` column of a 3D dataset, and seeing weird results because of this :)

### Steps for implementing the feature

- At the very least, [improving documentation ](https://aedt.docs.pyansys.com/version/0.9/API/_autosummary/pyaedt.application.Variables.DataSet.update.html) to reflect this sorting behavior would be helpful.
- Make the sorting optional, if possible through a passed argument.

Samuelopez-ansys, 2024-06-13:
Hi @nathmay, please could you provide a 3D Dataset just to check the new code is working as expected?

Thanks!

Samuelopez-ansys, 2024-06-13:
@nathmay  Please try: https://github.com/ansys/pyaedt/pull/4803

PR #4803 (2024-06-13): FEAT: Sort flag in dataset
Close #4801 

By default is sorting with X values, but in 3D Dataset could be dangerous, so we have added a disable of the sorting.
closed 2024-06-13 by maxcapodi78