ansys/pyaedt #4930: After opening 22R2, cannot open 24R1 or 24R2.

zebanaqviWAT, 2024-07-17:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

`from pyaedt import Hfss, settings

settings.use_grpc_api = True

with Hfss(version="2022.2", close_on_exit=True) as hfss:
  pass

with Hfss(version="2024.2", close_on_exit=True) as hfss:
  pass`

### Steps To Reproduce

Run the above script using pyaedt version 0.9.8 or 0.9.9.
You would need 22R2 and 24Rx installed.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

Note that python version is 3.11
(venv) C:\Users\znaqvi>python --version
Python 3.11.4

(venv) C:\Users\znaqvi>python -m pip freeze
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==23.2.0
certifi==2024.7.4
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
contourpy==1.2.1
cycler==0.12.1
defusedxml==0.7.1
fonttools==4.53.1
fpdf2==2.7.9
idna==3.7
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
matplotlib==3.9.1
numpy==1.26.4
packaging==24.1
pandas==2.2.2
pillow==10.4.0
platformdirs==4.2.2
plumbum==1.8.3
pooch==1.8.2
psutil==6.0.0
pyaedt==0.9.9
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.20.0
pyparsing==3.1.2
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.44.0
pywin32==306
referencing==0.35.1
requests==2.32.3
rpds-py==0.19.0
rpyc==6.0.0
Rtree==1.3.0
scooby==0.10.0
six==1.16.0
toml==0.10.2
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.2
vtk==9.3.1

gmalinve, 2024-07-18:
Hi @zebanaqviWAT,

you can now open two AEDT sessions with 2 different versions by leveraging the multi desktop functionality.
In the Settings class we have recently introduced a new property called _use_multi_desktop_ that allows you to have as many AEDT instances as you want (even with different versions).
Add this at the beginning of your code snippet:

`from pyaedt import settings`
`settings.use_multi_desktop = True`

and then you can open the sessions as:

`output_desktop = Desktop(new_desktop=True,version=241,non_graphical=non_graphical)`
`input_desktop = Desktop(new_desktop=True, version=222, non_graphical=non_graphical)`

and then you could release them separately:

`input_desktop.release_desktop()`
`output_desktop.release_desktop()`

I'll open a new discussion to cover this topic with a simple example.

Hope this helps,

Giulia

gmalinve, 2024-07-18:
here the discussion: https://github.com/ansys/pyaedt/discussions/4933

gmalinve, 2024-07-18:
@zebanaqviWAT I'll close this PR hoping that the explanation was clear enough for you.

Thanks,

Giulia

closed 2024-07-18 by gmalinve

zebanaqviWAT, 2024-07-24:
@gmalinve The AEDT sessions open in succession in my script - open 22.2, do stuff, close it; open 24.2, do stuff, close it. Do you know why opening in sequence is not working? I tried to use_multi_desktop but it got stuck at the launch of 24.2.

gmalinve, 2024-07-29:
Hi @zebanaqviWAT, with the example provided in the discussion you can open them sequentially. The process should not get stuck.
Are you setting grpc to true also?

zebanaqviWAT, 2024-08-01:
Hi @gmalinve, 
Yes, grpc is set to True.
The application of this script is to open versions using older projects that used ACIS kernel. Will the use_multi_desktop work with older versions?
It would be good to find the root cause for why 24.2 and 24.1 fail to launch but 23.2 does after launching 22.2 with the 'with Hfss(...):' method.

Samuelopez-ansys, 2024-08-01:
@zebanaqviWAT Yes it is possible.

please see this extension:

https://aedt.docs.pyansys.com/version/stable/User_guide/pyaedt_extensions_doc/project/kernel_convert.html

I think it is doing exactly what you are looking for. You could check the source code inside pyaedt/workflows/