ansys/pyaedt #4950: Bug located in ...SolutionData

wsp6, 2024-07-24:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The SolutionData.data_real function does not output the full matrix from an object containing gain data.  Only the first row of the matrix is output.

### Steps To Reproduce

        data=hfss.post.get_solution_data(
            expressions='GainTotal',
            setup_sweep_name='Setupname : Sweepname',
            variations=variationsdict[first_value],
            primary_sweep_variable='Phi',
            report_category="Far Fields",
            context="3D",
            math_formula='dB',
        )

       gain_real_values=data.data_real()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==23.2.0
bcrypt @ file:///D:/bld/bcrypt_1715971809558/work
beautifulsoup4==4.12.3
Bottleneck==1.4.0
Brotli==1.1.0
cachetools==5.4.0
certifi @ file:///C:/b/abs_7bbxxprxir/croot/certifi_1720453609121/work/certifi
cffi @ file:///C:/b/abs_78eb1_vq6z/croot/cffi_1714483206096/work
charset-normalizer @ file:///home/conda/feedstock_root/build_artifacts/charset-normalizer_1698833585322/work
clr-loader @ file:///home/conda/feedstock_root/build_artifacts/clr_loader_1691434961632/work
colorama @ file:///C:/b/abs_a9ozq0l032/croot/colorama_1672387194846/work
contourpy @ file:///D:/bld/contourpy_1712430020791/work
cryptography==42.0.8
cycler @ file:///home/conda/feedstock_root/build_artifacts/cycler_1696677705766/work
defusedxml @ file:///tmp/build/80754af9/defusedxml_1615228127516/work
fonttools==4.53.1
fpdf2 @ file:///home/conda/feedstock_root/build_artifacts/fpdf2_1716224807835/work
google==3.0.0
google-api-core==2.19.1
google-api-python-client==2.137.0
google-auth==2.32.0
google-auth-httplib2==0.2.0
google-auth-oauthlib==1.2.1
google-pasta==0.2.0
googleAPI==0.1.0
googleapis-common-protos==1.63.2
httplib2==0.22.0
idna @ file:///C:/b/abs_aad84bnnw5/croot/idna_1714398896795/work
importlib_resources @ file:///C:/b/abs_e9f8_0_2dq/croot/importlib_resources-suite_1720641112288/work
intel-cmplr-lib-ur==2024.2.0
intel-openmp==2024.2.0
jsonschema @ file:///C:/b/abs_d1c4sm8drk/croot/jsonschema_1699041668863/work
jsonschema-specifications==2023.12.1
kiwisolver @ file:///D:/bld/kiwisolver_1695380110085/work
loguru @ file:///D:/bld/loguru_1695547517194/work
matplotlib==3.9.1
mkl==2024.2.0
mkl-fft @ file:///C:/b/abs_19i1y8ykas/croot/mkl_fft_1695058226480/work
mkl-random @ file:///C:/b/abs_edwkj1_o69/croot/mkl_random_1695059866750/work
mkl-service==2.4.1
numexpr==2.10.1
numpy==1.26.4
oauth2client==4.1.3
oauthlib==3.2.2
opencv-python==4.10.0.84
packaging @ file:///C:/b/abs_c3vlh0z4jw/croot/packaging_1720101866539/work
pandas==2.2.2
paramiko @ file:///home/conda/feedstock_root/build_artifacts/paramiko_1703015906107/work
pillow @ file:///C:/b/abs_32o8er3uqp/croot/pillow_1721059447598/work
pip-review==1.3.0
platformdirs @ file:///home/conda/feedstock_root/build_artifacts/platformdirs_1715777629804/work
plumbum @ file:///home/conda/feedstock_root/build_artifacts/plumbum_1714378247975/work
ply==3.11
pooch==1.8.2
proto-plus==1.24.0
protobuf==5.27.2
psutil==6.0.0
pyaedt @ file:///D:/bld/pyaedt_1720910707818/work
pyasn1==0.6.0
pyasn1_modules==0.4.0
pybase64==1.3.2
pycparser @ file:///home/conda/feedstock_root/build_artifacts/pycparser_1711811537435/work
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.21.1
PyNaCl @ file:///C:/Windows/Temp/abs_d5c3ajcm87/croots/recipe/pynacl_1659620667490/work
pyparsing==3.1.2
PyQt5==5.15.11
PyQt5-Qt5==5.15.2
PyQt5_sip==12.15.0
PySocks @ file:///C:/ci/pysocks_1605307512533/work
python-dateutil @ file:///C:/b/abs_3au_koqnbs/croot/python-dateutil_1716495777160/work
pythonnet @ file:///home/conda/feedstock_root/build_artifacts/pythonnet_1697010695704/work
pytomlpp @ file:///D:/bld/pytomlpp_1695824633871/work
pytz @ file:///C:/b/abs_6ap4tsz1ox/croot/pytz_1713974360290/work
pyvista @ file:///home/conda/feedstock_root/build_artifacts/pyvista_1720470165236/work
pywin32==306
referencing==0.35.1
requests==2.32.3
requests-oauthlib==2.0.0
rpds-py==0.19.0
rpyc==6.0.0
rsa==4.9
Rtree==1.3.0
scooby @ file:///home/conda/feedstock_root/build_artifacts/scooby_1714897440316/work
sip==6.8.6
six @ file:///tmp/build/80754af9/six_1644875935023/work
soupsieve==2.5
tbb==2021.13.0
toml==0.10.2
tomli @ file:///C:/Windows/TEMP/abs_ac109f85-a7b3-4b4d-bcfd-52622eceddf0hy332ojo/croots/recipe/tomli_1657175513137/work
tornado==6.4.1
typing_extensions==4.12.2
tzdata @ file:///home/conda/feedstock_root/build_artifacts/python-tzdata_1707747584337/work
unicodedata2 @ file:///C:/b/abs_b6apldlg7y/croot/unicodedata2_1713212998255/work
uritemplate==4.1.1
urllib3 @ file:///C:/b/abs_a7hvzm4y95/croot/urllib3_1718912661242/work
vtk===-PKG-VERSION
waiting==1.4.1
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306162074/work
win32-setctime @ file:///home/tkoch/Workspace/win32_setctime/win32_setctime_1643630045199/work
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1718013267051/work

Samuelopez-ansys, 2024-07-26:
Hi @wsp6 ,

"data" is an object that contains a lot of information and properties, one property is:
data.full_matrix_mag_phase

which I think is what you need.

If you want to perform far-field post-processing, there are advanced methods:

https://aedt.docs.pyansys.com/version/stable/examples/02-HFSS/Array.html#get-far-field-data

I am refactoring it to have an even method class, but you can use it in the meanwhile.

wsp6, 2024-07-26:
Thanks for the help.  The full_matrix_mag_phase information is the absolute value of the gain.  I need to be able to differentiate between +1 dB and -1 dB because my peak gain values should be close to 0 dB. So I need the actual gain values.

The object "data" is exporting properly.  When I pause execution and look into memory, the object has the full two dimensional gain matrix in data > full_matrix_real_imag > position 0.  The problem is when I attempt to extract the data from the object using the included "data_real" function.  The "data_real" function only extracts a one dimensional list instead of the two dimensional matrix.

I have attempted to use the get-far-field-data method, it returns the error "No Data Available. Check inputs" with the following code:
       datafarfieldmethod=hfss.post.get_far_field_data(
            expressions='GainTotal',
            setup_sweep_name='Setupname : Sweepname',
            domain='3D',
        )
I know the data is there because it's in the object exported using the get_solution_data method.

As an alternative I've tried to pull RadiationEfficiency, PeakGain, and/or PeakRealizedGain from the 'Antenna Parameters' category.  I can't get any object to export from the 'Antenna Parameters' category. 'Antenna Parameters' solutions data won't export at all and any 'Antenna Parameters' reports generated using PyAedt are blank with no trace.

Samuelopez-ansys, 2024-08-12:
Hi @wsp6 ,

This is an expected behavior. The builtin methods data_real, data_magnitude... are one dimension. If you want to perform a specific operation of the matrix, you can do it afterwards.

If you know how to use pandas, I encourage you to convert this data to pandas format:

data.enable_pandas_output = True

You have one more alternative, last PyAEDT release introduce a new to export farfields, which is very powerful:

https://aedt.docs.pyansys.com/version/stable/API/Visualization.html#farfield

Please close this issue if you think it can be closed.

wsp6, 2024-08-12:
Thank you for the help. I'll check out pandas and the new farfield export option.  Again, thanks for the help.

closed 2024-08-12 by wsp6