ansys/pyaedt #4979: Bug located in ...The code to delete a point entity does not work

Herve-L1, 2024-07-31:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The code to delete a point entity does not work
Problem exist with Maxwell2024R2 and Maxwell2022R3

### Steps To Reproduce

[delete2024_pt.zip](https://github.com/user-attachments/files/16441521/delete2024_pt.zip)
In this file there is the 3D model and the python code which removes 2 entities named wk_pt. and another Rectangle1.
**running detail**

> *** Python 3.10.12 (remotes/origin/3b5d0b535dfa1177dc4fc3b6f40c990a1085eaab-dirty:3b5d0b53, Mar 25 2) [MSC v.1920 64 bit (AMD64)] on win32. ***
> >>> 
> *** Remote Interpreter Reinitialized ***
> PyAEDT WARNING: Argument `specified_version` is deprecated for method `__init__`; use `version` instead.
> PyAEDT INFO: Initializing new Desktop session.
> PyAEDT INFO: StdOut is enabled
> PyAEDT INFO: Log on file is enabled
> PyAEDT INFO: Log on Desktop Message Manager is enabled
> PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
> PyAEDT INFO: Launching PyAEDT outside AEDT with gRPC plugin.
> PyAEDT INFO: Found active AEDT gRPC session on port 50051
> PyAEDT INFO: AEDT installation Path C:\Program Files\AnsysEM\v242\Win64
> PyAEDT INFO: pyaedt v0.9.9
> PyAEDT INFO: Python version 3.10.12 (remotes/origin/3b5d0b535dfa1177dc4fc3b6f40c990a1085eaab-dirty:3b5d0b53, Mar 25 2) [MSC v.1920 64 bit (AMD64)]
> PyAEDT INFO: AEDT 2024.2.0 Build Date 2024-05-23 02:23:50
> PyAEDT INFO: No project is defined. Project basic_Box_test1 exists and has been read.
> PyAEDT INFO: Active Design set to Maxwell2DDesign1
> PyAEDT INFO: Aedt Objects correctly read
> PyAEDT INFO: Modeler2D class has been initialized!
> PyAEDT INFO: Modeler class has been initialized! Elapsed time: 0m 1sec
> PyAEDT WARNING: No objects to delete
> PyAEDT INFO: Parsing design objects. This operation can take time
> PyAEDT INFO: Parsing B:/AEDT_coding_training/Maxwell2024/basic_Box_test1.aedt.
> PyAEDT INFO: File B:/AEDT_coding_training/Maxwell2024/basic_Box_test1.aedt correctly loaded. Elapsed time: 0m 0sec
> PyAEDT INFO: aedt file load time 0.03402590751647949
> PyAEDT INFO: 3D Modeler objects parsed. Elapsed time: 0m 0sec
> PyAEDT INFO: Deleted 1 Objects: Rectangle1.
> PyAEDT INFO: Desktop has been released.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

Microsoft Windows [version 10.0.19045.4651]
(c) Microsoft Corporation. Tous droits réservés.

C:\Users\212508492>cd C:\Users\212508492\AppData\Roaming\.pyaedt_env\3_10\Scripts

C:\Users\212508492\AppData\Roaming\.pyaedt_env\3_10\Scripts>python -m pip freeze
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
anyio==4.4.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
Babel==2.15.0
beautifulsoup4==4.12.3
bleach==6.1.0
certifi==2024.7.4
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.2.1
cycler==0.12.1
debugpy==1.8.2
decorator==5.1.1
defusedxml==0.7.1
et-xmlfile==1.1.0
exceptiongroup==1.2.2
executing==2.0.1
fastjsonschema==2.20.0
fonttools==4.53.1
fpdf2==2.7.9
fqdn==1.5.1
geopandas==1.0.1
h11==0.14.0
httpcore==1.0.5
httpx==0.27.0
idna==3.7
imageio==2.34.2
ipycanvas==0.13.2
ipyevents==2.0.2
ipykernel==6.29.5
ipython==8.26.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.8.2
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.4
joblib==1.4.2
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyter_server==2.14.2
jupyter_server_terminals==0.5.3
jupyterlab==4.2.4
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
jupyterlab_widgets==1.1.8
kiwisolver==1.4.5
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.9.1
matplotlib-inline==0.1.7
mdurl==0.1.2
meshio==5.3.5
mistune==3.0.2
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
nest-asyncio==1.6.0
networkx==3.3
notebook==7.2.1
notebook_shim==0.2.4
numpy==1.26.4
openpyxl==3.1.5
osmnx==1.9.3
overrides==7.7.0
packaging==24.1
pandas==2.2.2
pandocfilters==1.5.1
parso==0.8.4
pillow==10.4.0
platformdirs==4.2.2
plumbum==1.8.3
pooch==1.8.2
prometheus_client==0.20.0
prompt_toolkit==3.0.47
psutil==6.0.0
pure_eval==0.2.3
pyaedt==0.9.9
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.21.1
Pygments==2.18.0
pyogrio==0.9.0
pyparsing==3.1.2
pyproj==3.6.1
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.44.1
pywin32==306
pywinpty==2.0.13
PyYAML==6.0.1
pyzmq==26.0.3
referencing==0.35.1
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.7.1
rpds-py==0.19.0
rpyc==6.0.0
Rtree==1.3.0
scikit-learn==1.5.1
scikit-rf==1.1.0
scipy==1.14.0
scooby==0.10.0
Send2Trash==1.8.3
shapely==2.0.5
six==1.16.0
sniffio==1.3.1
soupsieve==2.5
SRTM.py==0.3.7
stack-data==0.6.3
terminado==0.18.1
threadpoolctl==3.5.0
tinycss2==1.3.0
toml==0.10.2
tomli==2.0.1
tornado==6.4.1
traitlets==5.14.3
types-python-dateutil==2.9.0.20240316
typing_extensions==4.12.2
tzdata==2024.1
uri-template==1.3.0
urllib3==2.2.2
utm==0.7.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.6.0
webencodings==0.5.1
websocket-client==1.8.0
widgetsnbextension==3.6.7

C:\Users\212508492\AppData\Roaming\.pyaedt_env\3_10\Scripts>

gmalinve, 2024-08-01:
Hi @Herve-L1,

Thanks for opening this issue.
In your code snippet you are not using PyAEDT but rather still using the native API.
You are only using PyAEDT to create Maxwell 2d instance and to release the desktop.
You can delete your point with PyAEDT in an easier way.
1. Attach to your Maxwell 2d instance: `m2d = Maxwell2d(version="2024.2")`
2. Get the point object: `my_point = m2d.modeler.points["wk_pt"]`
3. Delete your point: `my_point.delete()`
4. Release desktop: `m2d.release_desktop(False, False)`

Hope this helps.
Kind regards,

Giulia

gmalinve, 2024-08-06:
Hi @Herve-L1,

Does this answer your question?
Please do let us know if you have further questions and we can proceed closing this issue.

Regards,

Giulia

closed 2024-08-08 by Samuelopez-ansys

Herve-L1, 2024-08-14:
[point test with following code.docx](https://github.com/user-attachments/files/16613196/point.test.with.following.code.docx)
I have made test with following code.

# -*- coding: utf-8 -*-
#################################################################################
import sys, os
import time
from pyaedt import Maxwell2d
#m2d = Maxwell2d(specified_version="2024.1")
m2d = Maxwell2d(specified_version="2022.2", close_on_exit=True)
oDesktop = m2d.odesktop
oProject = oDesktop.GetActiveProject()
oDesign = oProject.GetActiveDesign()
MyProject = oProject.GetName()
Mydesign = oDesign.GetName()
oEditor = m2d.modeler
Global_start_time = time.time()
m2d.modeler.points["wk_pt"].delete()
Global_temps_load = ("Runing Time : %s secondes ---" % (time.time() - Global_start_time))
print (Global_temps_load)
m2d.release_desktop(close_projects=False, close_desktop=False)
a=1

*** Python 3.7.13 (remotes/origin/bdighole/python/3713/clean_up-dirty:d4e1bece, Apr  6 2022, 00:24:) [MSC v.1920 64 bit (AMD64)] on win32. ***
*** Distant Python engine is active ***
>>> 
*** Remote Interpreter Reinitialized ***
PyAEDT INFO: Initializing new Desktop session.
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Log on file is enabled
PyAEDT INFO: Log on Desktop Message Manager is enabled
PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
PyAEDT INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
PyAEDT INFO: AEDT installation Path C:\Program Files\AnsysEM\v222\Win64.
PyAEDT INFO: AEDT 2022.2 Started with process ID 21252.
PyAEDT INFO: pyaedt v0.9.1
PyAEDT INFO: Python version 3.7.13 (remotes/origin/bdighole/python/3713/clean_up-dirty:d4e1bece, Apr  6 2022, 00:24:) [MSC v.1920 64 bit (AMD64)]
PyAEDT INFO: AEDT 2022.2.3 Build Date 2022-09-22 14:19:20
PyAEDT INFO: No project is defined. Project basic_Box_test1 exists and has been read.
PyAEDT INFO: Active Design set to Maxwell2DDesign1
PyAEDT INFO: Aedt Objects correctly read
PyAEDT INFO: Parsing B:/AEDT_coding_training/Maxwell2022/basic_Box_test1.aedt.
PyAEDT INFO: File B:/AEDT_coding_training/Maxwell2022/basic_Box_test1.aedt correctly loaded. Elapsed time: 0m 0sec
PyAEDT INFO: aedt file load time 0.040021419525146484
PyAEDT INFO: Modeler2D class has been initialized!
PyAEDT INFO: Modeler class has been initialized! Elapsed time: 0m 0sec
Runing Time : 0.011139631271362305 secondes ---
PyAEDT INFO: Desktop has been released.
>>> 

results are correct.
But if there is no point program crash.
For the other delete object rectangle as example, if there is no rectangle AEDT manage the error without crash.
Also, for point entity there is no method for move.
 
Move could be made by changing Position value, but position property is not readable and could not be set.
In my beginning of programming, I have made a mistake with install and have Maxwell 2022.2.3 with AEDT v0.9.1
In this configuration” delete("wk_pt")” run correctly
Here running detail:
# -*- coding: utf-8 -*-
#################################################################################
import sys, os
import time

from pyaedt import Maxwell2d
#m2d = Maxwell2d(specified_version="2024.1")
m2d = Maxwell2d(specified_version="2022.2", close_on_exit=True)
oDesktop = m2d.odesktop
oProject = oDesktop.GetActiveProject()
oDesign = oProject.GetActiveDesign()
MyProject = oProject.GetName()
Mydesign = oDesign.GetName()
oEditor = m2d.modeler
Global_start_time = time.time()
oEditor.delete("wk_pt")
Global_temps_load = ("Runing Time : %s secondes ---" % (time.time() - Global_start_time))
print (Global_temps_load)
m2d.release_desktop(close_projects=False, close_desktop=False)
a=1

*** Python 3.7.13 (remotes/origin/bdighole/python/3713/clean_up-dirty:d4e1bece, Apr  6 2022, 00:24:) [MSC v.1920 64 bit (AMD64)] on win32. ***
>>> 
*** Remote Interpreter Reinitialized ***
PyAEDT INFO: Initializing new Desktop session.
PyAEDT INFO: StdOut is enabled
PyAEDT INFO: Log on file is enabled
PyAEDT INFO: Log on Desktop Message Manager is enabled
PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
PyAEDT INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
PyAEDT INFO: AEDT installation Path C:\Program Files\AnsysEM\v222\Win64.
PyAEDT INFO: AEDT 2022.2 Started with process ID 18908.
PyAEDT INFO: pyaedt v0.9.1
PyAEDT INFO: Python version 3.7.13 (remotes/origin/bdighole/python/3713/clean_up-dirty:d4e1bece, Apr  6 2022, 00:24:) [MSC v.1920 64 bit (AMD64)]
PyAEDT INFO: AEDT 2022.2.3 Build Date 2022-09-22 14:19:20
PyAEDT INFO: No project is defined. Project basic_Box_test1 exists and has been read.
PyAEDT INFO: Active Design set to Maxwell2DDesign1
PyAEDT INFO: Aedt Objects correctly read
PyAEDT INFO: Parsing B:/AEDT_coding_training/Maxwell2022/basic_Box_test1.aedt.
PyAEDT INFO: File B:/AEDT_coding_training/Maxwell2022/basic_Box_test1.aedt correctly loaded. Elapsed time: 0m 0sec
PyAEDT INFO: aedt file load time 0.03298616409301758
PyAEDT INFO: Modeler2D class has been initialized!
PyAEDT INFO: Modeler class has been initialized! Elapsed time: 0m 0sec
PyAEDT INFO: Deleted 1 Objects: wk_pt.
Runing Time : 0.04803204536437988 secondes ---
PyAEDT INFO: Desktop has been released.

Could improve functionality for point in delete and move function?