ansys/pyaedt #4993: pyAEDT: License remains in use after program crash

pasc1377, 2024-08-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am currently developing a script for the Ansys Electronics Desktop using pyAEDT.
From time to time, errors occur in the programme, resulting in incorrect programme closures because the code is not executed completely.
As a result, the EDT licence remains in use and I cannot start any more tests. Only a restart of the licence server solves the problem.
Unfortunately, this is quite inconvenient as the server is running on another PC and the restart is quite time-consuming.
Are you aware of this problem or do you have an idea for a workaround?
Is it possible to restart the licence server via the command line from another computer (remote)?

### Steps To Reproduce

1. Create Maxwell2d object
2. Create modeler object
3. Doing some experiments 
4. Program crash
5. License remains in use

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==23.2.0
certifi==2024.7.4
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
contourpy==1.2.1
cycler==0.12.1
defusedxml==0.7.1
fonttools==4.53.1
fpdf2==2.7.9
idna==3.7
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
matplotlib==3.9.1
numpy==1.26.4
packaging==24.1
pandas==2.2.2
pillow==10.4.0
platformdirs==4.2.2
plumbum==1.8.3
pooch==1.8.2
psutil==6.0.0
pyaedt==0.9.10
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.23.0
pyparsing==3.1.2
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.44.1
pywin32==306
referencing==0.35.1
requests==2.32.3
rpds-py==0.19.1
rpyc==6.0.0
Rtree==1.3.0
scikit-rf==1.2.0
scipy==1.14.0
scooby==0.10.0
six==1.16.0
toml==0.10.2
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.2
vtk==9.3.1

Samuelopez-ansys, 2024-08-05:
Hi @pasc1377 ,

This is a known issue, but from PyAEDT there is no way to fix it because if the process is not released correctly, it is an expected behavior to remain it open. As a best practice, you should always release the PyAEDT object at the end of your script:

`app.release_desktop()`

If the script crashes you could also improve this performance by encapsulating your code inside a "with":

```
from pyaedt import Desktop, Circuit
with Desktop(version="2024.1", non_graphical=False, new_desktop_session=True, close_on_exit=True) as d:
        circuit = Circuit()
        d.logger.info("Successfully updated project definitions")
        circuit .save_project()
```

If the previous workaround does not work for you, the only solution is going to the task manager, and kill the AEDT process, this should release the license.

Regarding your last question, it is not related to PyAEDT, I am not aware of any way to restart the license manager from the command line, please submit a support case with this question.

In PyAEDT you have one method to check if there are available license before running your workflow.

```
from pyaedt.generic.general_methods import is_license_feature_available

license_available = is_license_feature_available()
```

THis method works when your local machine has access to the license server.

pasc1377, 2024-08-05:
Hi @Samuelopez-ansys,

thank you very much! That will help me a lot. Unfortunately, the process is not displayed at all in the Task Manager. So unfortunately I can't kill the process there either.

Samuelopez-ansys, 2024-08-05:
Hi @pasc1377 ,

So for now, there is not other approach, because if the script crashes, the connection is lost and the AEDT session can remain.

If you use 2024R1, the communication protocol has changed to GRPC, which should be more stable, but you could find similar issues if the script crashes, in that case, I suggest you additional best practices in your script, like saving the process ID and grpc port of the AEDT session, and then you could implement a "kill task", one way, for example, could be using pyaedt.Desktop,

you could reconnect to the "zombie" AEDT and perform a desktop.release_desktop(), which should close AEDT. Of course, you could use other python approaches to kill the specific process.

pasc1377, 2024-08-05:
Hi @Samuelopez-ansys,

Thank you very much. I will try it out. How can I read out the process ID and the GRPC port of the session? I have not yet found a possibility in the documentation.

Samuelopez-ansys, 2024-08-05:
@pasc1377 If you have a Maxwell object called m2d:

m2d.desktop_class.port
m2d.desktop_class.aedt_process_id

Both are properties of Desktop class.

pasc1377, 2024-08-05:
Tanke you so much!

closed 2024-08-05 by Samuelopez-ansys