ansys/pyaedt #4994: Failed to execute gRPC AEDT command: CreatePolyline

pasc1377, 2024-08-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I would like to use the oEditor in the conventional way and create a geometry. I have created a minimal example for this. Unfortunately, it does not work because there is an error when executing the command. Do you have any idea why this might be?

### Steps To Reproduce

Here the Minimal:
`import sys
from pyaedt import Maxwell2d

sys.path.append(r"C:\Program Files\AnsysEM\v241\Win64\PythonFiles\DesktopPlugin")
import ScriptEnv

aedt_version = "2024.1"

with Maxwell2d(project='Test', non_graphical=False, version=aedt_version, new_desktop=True, close_on_exit=True) as m2d:
    #ScriptEnv.Initialize("", False, "", 59522) # Port?
    oEditor = m2d.oeditor
    oEditor.CreatePolyline(
        [
            "NAME:PolylineParameters",
            "IsPolylineCovered:=", True,
            "IsPolylineClosed:=", False,
            [
                "NAME:PolylinePoints",
                [
                    "NAME:PLPoint",
                    "X:="	, "0mm",
                    "Y:="			, "0mm",
                    "Z:="			, "0mm"
                ],
                [
                    "NAME:PLPoint",
                    "X:="			, "1.3mm",
                    "Y:="			, "1.1mm",
                    "Z:="			, "0mm"
                ]
            ],
            [
                "NAME:PolylineSegments",
                [
                    "NAME:PLSegment",
                    "SegmentType:="		, "Line",
                    "StartIndex:="		, 0,
                    "NoOfPoints:="		, 2
                ]
            ],
            [
                "NAME:PolylineXSection",
                "XSectionType:"	, "None",
                "XSectionOrient:"	, "Auto",
                "XSectionWidth:"	, "0mm",
                "XSectionTopWidth:"	, "0mm",
                "XSectionHeight:"	, "0mm",
                "XSectionNumSegments:"	, "0",
                "XSectionBendType:"	, "Corner"
            ]
        ],
        [
            "NAME:Attributes",
            "Name:="		, "Polyline1",
            "Flags:="		, "",
            "Color:="		, "(143 175 143)",
            "Transparency:"	, 0,
            "PartCoordinateSystem:=", "Global",
            "UDMId:="		, "",
            "MaterialValue:"	, "\"vacuum\"",
            "SurfaceMaterialValue:=", "\"\"",
            "SolveInside:="		, True,
            "ShellElement:"	, False,
            "ShellElementThickness:=", "0mm",
            "ReferenceTemperature:=", "20cel",
            "IsMaterialEditable:"	, True,
            "UseMaterialAppearance:=", False,
            "IsLightweight:"	, False
        ])

    m2d.save_project(r"C:\Users\schw_p6\Documents\Python\IAS_Electric_Machine_Toolkit\Test\TestMinimal.aedt")
    m2d.close_desktop()

print('Finished')
`

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==23.2.0
certifi==2024.7.4
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
contourpy==1.2.1
cycler==0.12.1
defusedxml==0.7.1
fonttools==4.53.1
fpdf2==2.7.9
idna==3.7
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
matplotlib==3.9.1
numpy==1.26.4
packaging==24.1
pandas==2.2.2
pillow==10.4.0
platformdirs==4.2.2
plumbum==1.8.3
pooch==1.8.2
psutil==6.0.0
pyaedt==0.9.10
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.23.0
pyparsing==3.1.2
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.44.1
pywin32==306
referencing==0.35.1
requests==2.32.3
rpds-py==0.19.1
rpyc==6.0.0
Rtree==1.3.0
scikit-rf==1.2.0
scipy==1.14.0
scooby==0.10.0
six==1.16.0
toml==0.10.2
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.2
vtk==9.3.1

Samuelopez-ansys, 2024-08-05:
Hi @pasc1377 ,

First, why are you using the AEDT API instead of PyAEDT? It is much easier.

line = m2d.modeler.create_polyline(points=....)

https://aedt.docs.pyansys.com/version/stable/API/_autosummary/pyaedt.modeler.modeler2d.Modeler2D.create_polyline.html

About your code, I think the polyline is not correct, I am trying to create it using the IronPython console, but the point definition is wrong, please record the script again, and try to re run the code to verify the polyline can be created.

pasc1377, 2024-08-05:
Hello,

There are two reasons for this:
1) I have a lot of parameterised geometries in the "old design" and want to convert them to pyAEDT piece by piece.
2) The much more important point: I have done a few tests. Building the model with pyAEDT (2D electromagnetic) took about 1.5 times as long as with the previous version. For this reason, I thought that there could be time advantages if I did not use pyAEDT commands, especially for geometry creation.
Have you observed anything similar?

When I insert a Maxwell2D design and click on "Tools-->Run Script...", the creation works.
Here is the recorded script:
 
`# ----------------------------------------------
# Script Recorded by Ansys Electronics Desktop Version 2024.1.0
# 18:03:40  Aug 01, 2024
# ----------------------------------------------
import ScriptEnv
ScriptEnv.Initialize("Ansoft.ElectronicsDesktop")
oDesktop.RestoreWindow()
oProject = oDesktop.SetActiveProject("Project2")
oDesign = oProject.SetActiveDesign("Maxwell2DDesign1")
oEditor = oDesign.SetActiveEditor("3D Modeler")
oEditor.CreatePolyline(
	[
		"NAME:PolylineParameters",
		"IsPolylineCovered:="	, True,
		"IsPolylineClosed:="	, False,
		[
			"NAME:PolylinePoints",
			[
				"NAME:PLPoint",
				"X:="			, "0mm",
				"Y:="			, "0mm",
				"Z:="			, "0mm"
			],
			[
				"NAME:PLPoint",
				"X:="			, "1.3mm",
				"Y:="			, "1.1mm",
				"Z:="			, "0mm"
			]
		],
		[
			"NAME:PolylineSegments",
			[
				"NAME:PLSegment",
				"SegmentType:="		, "Line",
				"StartIndex:="		, 0,
				"NoOfPoints:="		, 2
			]
		],
		[
			"NAME:PolylineXSection",
			"XSectionType:="	, "None",
			"XSectionOrient:="	, "Auto",
			"XSectionWidth:="	, "0mm",
			"XSectionTopWidth:="	, "0mm",
			"XSectionHeight:="	, "0mm",
			"XSectionNumSegments:="	, "0",
			"XSectionBendType:="	, "Corner"
		]
	], 
	[
		"NAME:Attributes",
		"Name:="		, "Polyline1",
		"Flags:="		, "",
		"Color:="		, "(143 175 143)",
		"Transparency:="	, 0,
		"PartCoordinateSystem:=", "Global",
		"UDMId:="		, "",
		"MaterialValue:="	, "\"vacuum\"",
		"SurfaceMaterialValue:=", "\"\"",
		"SolveInside:="		, True,
		"ShellElement:="	, False,
		"ShellElementThickness:=", "0mm",
		"ReferenceTemperature:=", "20cel",
		"IsMaterialEditable:="	, True,
		"UseMaterialAppearance:=", False,
		"IsLightweight:="	, False
	])
`

Samuelopez-ansys, 2024-08-05:
Hi @pasc1377 ,

This code works for me:

import pyaedt

app = pyaedt.Maxwell2d(version="2024.1")

oEditor = app.oeditor
oEditor.CreatePolyline(
    [
        "NAME:PolylineParameters",
        "IsPolylineCovered:=", True,
        "IsPolylineClosed:=", False,
        [
            "NAME:PolylinePoints",
            [
                "NAME:PLPoint",
                "X:=", "0mm",
                "Y:=", "0mm",
                "Z:=", "0mm"
            ],
            [
                "NAME:PLPoint",
                "X:=", "1.3mm",
                "Y:=", "1.1mm",
                "Z:=", "0mm"
            ]
        ],
        [
            "NAME:PolylineSegments",
            [
                "NAME:PLSegment",
                "SegmentType:=", "Line",
                "StartIndex:=", 0,
                "NoOfPoints:=", 2
            ]
        ],
        [
            "NAME:PolylineXSection",
            "XSectionType:=", "None",
            "XSectionOrient:=", "Auto",
            "XSectionWidth:=", "0mm",
            "XSectionTopWidth:=", "0mm",
            "XSectionHeight:=", "0mm",
            "XSectionNumSegments:=", "0",
            "XSectionBendType:=", "Corner"
        ]
    ],
    [
        "NAME:Attributes",
        "Name:=", "Polyline1",
        "Flags:=", "",
        "Color:=", "(143 175 143)",
        "Transparency:=", 0,
        "PartCoordinateSystem:=", "Global",
        "UDMId:=", "",
        "MaterialValue:=", "\"vacuum\"",
        "SurfaceMaterialValue:=", "\"\"",
        "SolveInside:=", True,
        "ShellElement:=", False,
        "ShellElementThickness:=", "0mm",
        "ReferenceTemperature:=", "20cel",
        "IsMaterialEditable:=", True,
        "UseMaterialAppearance:=", False,
        "IsLightweight:=", False
    ])

Regarding time, I did not notice it. But PyAEDT creates a python object for each line, which gives you full control of each component, but AEDT API, does not give it you this python object:

line = app.modeler.create_polyline(points=[["0mm", "0mm", "0mm"], ["1.3mm", "1.1mm", "0mm"]])

For that reason, it could take more time sometimes. I did a test creating 200 lines using both approaches, and this is the difference:

[image]

This is the code I used:

```
import pyaedt

app = pyaedt.Maxwell2d(version="2024.1")

app.logger.reset_timer()
for cont in range(1, 200):
    line = app.modeler.create_polyline(points=[["0mm", "0mm", "0mm"], ["1.3mm", "1.1mm", "0mm"]])
app.logger.info_timer("Line created with PyAEDT")

app.logger.reset_timer()
oEditor = app.oeditor
for cont in range(1, 200):
    oEditor.CreatePolyline(
        [
            "NAME:PolylineParameters",
            "IsPolylineCovered:=", True,
            "IsPolylineClosed:=", False,
            [
                "NAME:PolylinePoints",
                [
                    "NAME:PLPoint",
                    "X:=", "0mm",
                    "Y:=", "0mm",
                    "Z:=", "0mm"
                ],
                [
                    "NAME:PLPoint",
                    "X:=", "1.3mm",
                    "Y:=", "1.1mm",
                    "Z:=", "0mm"
                ]
            ],
            [
                "NAME:PolylineSegments",
                [
                    "NAME:PLSegment",
                    "SegmentType:=", "Line",
                    "StartIndex:=", 0,
                    "NoOfPoints:=", 2
                ]
            ],
            [
                "NAME:PolylineXSection",
                "XSectionType:=", "None",
                "XSectionOrient:=", "Auto",
                "XSectionWidth:=", "0mm",
                "XSectionTopWidth:=", "0mm",
                "XSectionHeight:=", "0mm",
                "XSectionNumSegments:=", "0",
                "XSectionBendType:=", "Corner"
            ]
        ],
        [
            "NAME:Attributes",
            "Name:=", "Polyline1",
            "Flags:=", "",
            "Color:=", "(143 175 143)",
            "Transparency:=", 0,
            "PartCoordinateSystem:=", "Global",
            "UDMId:=", "",
            "MaterialValue:=", "\"vacuum\"",
            "SurfaceMaterialValue:=", "\"\"",
            "SolveInside:=", True,
            "ShellElement:=", False,
            "ShellElementThickness:=", "0mm",
            "ReferenceTemperature:=", "20cel",
            "IsMaterialEditable:=", True,
            "UseMaterialAppearance:=", False,
            "IsLightweight:=", False
        ])
app.logger.info_timer("Line created with AEDT API")
```

pasc1377, 2024-08-05:
Hi @Samuelopez-ansys,

Thank you! Now I can see what the problem was.
And thanks also for explaining the reason for the time differences. If it were only a matter of readability and ease of programming, pyAEDT is unbeatable. It does make a difference in terms of time. A model setup currently takes about 1 minute and 10 seconds. With pyAEDT perhaps around 2 minutes.

closed 2024-08-05 by Samuelopez-ansys