ansys/pyaedt #5022: Cannot run Icepak toolkits after installing PyAEDT (EDT 24R2)

kruglovIFX, 2024-08-08:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Running any Icepak toolkit after installing pyaedt leads to an error pointing to the `import <ToolKit_name>`  line of the corresponding python script

[image]
[image]
[image]

### Steps To Reproduce

1. Install PyAEDT via "[PyAEDT Installer Python file]" as described in [Installation Instructions](https://aedt.docs.pyansys.com/version/stable/Getting_started/Installation.html)
2. Try any of the Icepak toolkits

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

-

Samuelopez-ansys, 2024-08-08:
Hi @kruglovIFX ,

I am not sure what you are trying to do. In your screenshot I see an Ironpython code, but this is PyAEDT GitHub issues, so it is not a PyAEDT issue.

Just for testing, I tried locally, and it worked for me:

[image]

Please submit a support case in the Ansys customer portal, this has to be related to the installation or any other IT issue, but again this should not be related to PyAEDT if you followed the installation guidelines as you mentioned. 

Did you install PyAEDT on syslib instead of PersonalLib?

kruglovIFX, 2024-08-08:
Hi @Samuelopez-ansys , thanks for your quick reply!

I've installed PyAEDT from the installer I downloaded from the website, it didn't ask me where to install it
[image]
I see that it was installed in `C:\Users\<user>\AppData\Roaming\.pyaedt_env`

The reason I am submitting it here because we see the issue with toolkits after installing PyAEDT this way. What we are trying to do - making use of Icepak Toolkits in EDT after installing pyaedt.

Samuelopez-ansys, 2024-08-08:
@kruglovIFX Indeed, but PyAEDT should not interact with AEDT, if you delete this folder (.pyaedt_env) does it work?

Later also try to delete your PersonalLib from the "Toolkits" folder. These are the only 2 places where the installer is interacting with your system. 

I think you should submit a support case, because this is a product issue.

lorenzovecchietti, 2024-08-11:
Hi @kruglovIFX,

last time I saw this behavior, it was due to Powershell 7 installed on the machine. Please try to uninstall it. 

If the problem persist, as @Samuelopez-ansys suggested, please open a support case in the customer portal.

kruglovIFX, 2024-08-19:
Hi @lorenzovecchietti ,

Unfortunately we cannot uninstall powershell on our working statins as we need it for certain things. Any workaround possible?

Thanks!

lorenzovecchietti, 2024-08-19:
If I were you, I would adjust the order of the environment variables so that the PATH for PowerShell 7 has a lower priority. But I've not tested this.

Since this issue relates to the core product, I recommend opening a ticket on the customer portal. This way, someone from the support team can assist you more effectively in finding and testing a suitable workaround.

Thank you!

closed 2024-08-19 by lorenzovecchietti