ansys/pyaedt #5023: Rework or clarify the way we use settings

SMoraisAnsys, 2024-08-08:
There seem to be some misunderstanding with how the code base works and how we want the final result to be.

This observation comes from @Alberto-DM which noticed that even when testing and setting the `pyaedt.settings` to not log into a file, we end up with a file being written. This comes from how python works. For example, in the following code extracted from our conftest.py file

```python
from pyaedt.generic.settings import settings

settings.enable_local_log_file = False
settings.enable_global_log_file = False
settings.number_of_grpc_api_retries = 6
settings.retry_n_times_time_interval = 0.5
settings.enable_error_handler = False
settings.enable_desktop_logs = False
settings.desktop_launch_timeout = 180
settings.release_on_exception = False
```

To import `settings` we are loading `pyaedt.__init__.py`, then `pyaedt.generic.__init__.py` and then `pyaedt.generic.settings.py`.
However, doing so leads instantiating `settings = Settings()` since other modules loaded through `pyaedt.__init__.py` do so.
Since the default value of `enable_global_log_file` is `True` then, if for whatever reason one of the file triggers some logging - like logging that one has not `pyvista`, `numpy`, or whateer - we'll have a file with the logged content, while the user might think that by setting `settings.enable_global_log_file = False` it prevented it.

I'm not sure of the correct approach to handle this as making changes might break users code or might require deep changes.

I'll investigate on the coming days and let you know what approach I can think about. Of course, any feedback is much appreciated !

Samuelopez-ansys, 2024-08-29:
@SMoraisAnsys Could we close this issue right?

SMoraisAnsys, 2024-08-29:
Closed by #5092

closed 2024-08-29 by SMoraisAnsys