ansys/pyaedt #5071: Add SetActiveDesign to the PyAEDT API

alevIFX, 2024-08-20:
### Description of the feature

Currently the SetActiveDesign(<designName>) is accessible via **oproject** legacy. 
Any chance to make it available in the PyAEDT API?

### Steps for implementing the feature

To reproduce the current status:

circuitapp = Circuit(project=projectName, design=designName)
circuitapp.oproject.SetActiveDesign(targetDesignName)

gmalinve, 2024-08-20:
Hi @alevIFX ,

This is already available in PyAEDT through: set_active_design(design_name).
For example: hfss.set_active_design(hfss.design_name)

Hope this helps,

Giulia

alevIFX, 2024-08-20:
I meant for Circuit and Netlist design, not HFSS design.
If you instantiate a Circuit() class for a Circuit Design and try to make a Netlist Design active then the Circuit.set_active_design(<netlistDesignName>) will not work.

alevIFX, 2024-08-20:
For completeness, the idea is to switch between different designs (and design types).
I've found that

desktop.active_design(name=<designName>) 

(where desktop is the object of the Desktop class) works and with this object also the GetModule function is available.

Look likes that the design activation shall be executed from the desktop object and not from the tool class.

Unfortunately the design name is retrieved differently depending if the design is a netlist or circuit design since the related function is not called nor working the same way.

closed 2024-08-29 by Samuelopez-ansys