ansys/pyaedt #5085: Bug located in Maxwell Examples

DaveTwyman, 2024-08-22:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

https://aedt.docs.pyansys.com/version/stable/examples/03-Maxwell/Maxwell3D_Team3_bath_plate.html#set-up-sweep-value-and-plot-solution

[image]

### Steps To Reproduce

Click on example here
https://aedt.docs.pyansys.com/version/stable/examples/03-Maxwell/Maxwell3D_Team3_bath_plate.html#set-up-sweep-value-and-plot-solution

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

Issue is on example website so this isn't needed

SMoraisAnsys, 2024-08-22:
Thanks @DaveTwyman, we have discovered this error (see https://github.com/ansys/pyaedt/issues/5082) and are working on it.

DaveTwyman, 2024-08-22:
@SMoraisAnsys , Thanks that's great

PR #5090 (2024-08-23): FIX: Solve setup in optimetrics
Close #5082 
Close #5085
closed 2024-08-23 by Samuelopez-ansys