ansys/pyaedt #5109: Add design notes

soliton37, 2024-08-30:
### Description of the feature

Develop a pyaedt command for adding a note to a design. 
Actually, I use this command line: hfss.odesign.editnotes(notes)
notes is a string

Samuelopez-ansys, 2024-08-30:
Hi @soliton37 , sure we can do it. But maybe you could consider doing your first contribution to PyAEDT :), everyone is welcome to contribute:

https://aedt.docs.pyansys.com/version/stable/Getting_started/Contributing.html

gmalinve, 2024-10-10:
Hi @soliton37,

This has been tackled in this PR: #5271, it's still open but soon to be closed and available in next release.
Please try to have a look at it so you can have some hints on how you could contribute to PyAEDT.

Thanks,

Regards,

Giulia

gmalinve, 2024-10-10:
Duplicate of #4613.

closed 2024-10-11 by anur7