ansys/pyaedt #5123: Bug located in ...PyAEDT Examples

DaveTwyman, 2024-09-03:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Two of the PyAEDT examples have the red 'Broken' stamp on them

https://aedt.docs.pyansys.com/version/stable/examples/04-Icepak/Icepak_CSV_Import.html#download-and-open-project

and 

https://aedt.docs.pyansys.com/version/stable/examples/01-HFSS3DLayout/EDB_in_3DLayout.html#initialize-aedt-and-launch-hfss-3d-layout

### Steps To Reproduce

Follow the links above

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

NA

gmalinve, 2024-10-01:
@Samuelopez-ansys @SMoraisAnsys can we close this issue? it doesn't make sense now that we are about to publish the new examples.

lorenzovecchietti, 2024-10-01:
Hi @DaveTwyman, I cannot see any error right now in those two examples. A recent PR might've solved those.

lorenzovecchietti, 2024-10-01:
Sorry @gmalinve, I was replying at the same time as you :)

gmalinve, 2024-10-01:
@lorenzovecchietti no worries :)

closed 2024-10-01 by gmalinve