ansys/pyaedt #5135: (AO) edb.stackup.load(stackup_file, rename=True) makes ignore original Anti PAD definition in VIA padstack

aohta2021, 2024-09-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

edb.configuration.load(config_file=file_json)
edb.configuration.run()
and also
edb.stackup.load(stackup_file, rename=True)

Both functions makes ignore anti pad definition in Padstack.
See:
[image]
[TEST_0905.zip](https://github.com/user-attachments/files/16883405/TEST_0905.zip)

### Steps To Reproduce

Please try to check attached script
1. Clear.bat
2. run_EDB.bat or run_XML.bat

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

(venv310) C:\Users\aohta\OneDrive - ANSYS, Inc\Documents\2024_BMT\Sony\Auto\To_Be_0904>python -m pip freeze
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==24.2.0
certifi==2024.8.30
cffi==1.17.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.3.0
cycler==0.12.1
defusedxml==0.7.1
et-xmlfile==1.1.0
fast_simplification==0.1.7
fiona==1.9.6
fonttools==4.53.1
fpdf2==2.7.9
geopandas==0.14.4
idna==3.8
imageio==2.35.1
joblib==1.4.2
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
markdown-it-py==3.0.0
matplotlib==3.9.2
mdurl==0.1.2
meshio==5.3.5
networkx==3.3
numpy==1.26.4
openpyxl==3.1.5
osmnx==1.9.4
packaging==24.1
pandas==2.2.2
pillow==10.4.0
platformdirs==4.2.2
plumbum==1.8.3
pooch==1.8.2
psutil==6.0.0
pyaedt==0.10.1
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.26.1
Pygments==2.18.0
pyparsing==3.1.4
pyproj==3.6.1
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.44.1
pywin32==306
PyYAML==6.0.2
referencing==0.35.1
requests==2.32.3
rich==13.8.0
rpds-py==0.20.0
rpyc==6.0.0
Rtree==1.3.0
scikit-learn==1.5.1
scikit-rf==1.2.0
scipy==1.14.1
scooby==0.10.0
shapely==2.0.6
six==1.16.0
SRTM.py==0.3.7
threadpoolctl==3.5.0
toml==0.10.2
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.2
utm==0.7.0
vtk==9.3.1

aohta2021, 2024-09-06:
@ring630 Please check it

aohta2021, 2024-09-10:
It is not only ignored Anti PAD definition but also Thermal definition.
It makes many DC shot errors.
[image]

If PyEDB give up this function, we could not set layer definition.
It is big impact to use PyEDB.

ring630, 2024-09-12:
@Samuelopez-ansys please help to move to pyedb