ansys/pyaedt #5137: Bug located in ... q3d.export_matrix_data()

gene820715, 2024-09-05:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

In 24R2 Q3D, the call to native API ExportMatrixData is missing \<pass\> argument, so the arguments after that are set incorrectly.

[image]

### Steps To Reproduce

1. q3d = Q3d(...)  #Open a q3d project with DCRL and ACRL solutions 
2. q3d.export_matrix_data("results.csv", "DC RL, AC RL", matrix_type="Maxwell,Spice,Couple", export_ac_dc_res=False, precision=5, field_width=8, use_sci_notation=False)
3. You will see in the csv ac_dc_res is there and precision is 8. Arguments are shifted.

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.8

### Installed packages

pyaedt and dependent packages

Samuelopez-ansys, 2024-09-09:
Hi @gene820715 ,

The PyAEDT method export_matrix_data is wrapping the AEDT API method ExportMatrixData:

https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/q3d.py#L424

I think the values passed to the AEDT API methods are correct, maybe you missed passing any of them. PyAEDT methods try to reduce the number of needed arguments if possible, and you probably pass the same order as the AEDT API method, but the PyAEDT method has different orders.

I encourage you to use the best practice of passing the name of the argument, then the order does not matter:

export_matrix_data(file_name=...., problem_type=...., variations=..., setup=..., ...)

Attached is the documentation:

https://aedt.docs.pyansys.com/version/stable/API/_autosummary/ansys.aedt.core.q3d.Q3d.export_matrix_data.html

Samuelopez-ansys, 2024-09-10:
@gene820715 You are right.

Let me investigate what this parameter is doing, and I will create a new pull request.

PR #5151 (2024-09-10): FIX: Add missed parameter "pass"
Close #5137
closed 2024-09-10 by Samuelopez-ansys