ansys/pyaedt #5153: PyAEDT can not import dxf drawing in Q3D

akselchang, 2024-09-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

PyAEDT can not import dxf drawing in Q3d design.
The error message as below picture

[image]

### Steps To Reproduce

1. code block as below:

```
import os
from pyaedt import Desktop, Q3d

project_name = 'test_project'
design_name = 'test_design'
desktop_version = '2024.1'
project_path_name = os.path.join(os.getcwd(), 'model', project_name + '.aedt')
dxf_name = 'Drawing1'
dxf_path_name = os.path.join(os.getcwd(), 'model', dxf_name + '.dxf')

# create q3d design
odesign = Q3d(project=project_name, design=design_name, version=desktop_version)

# set unit to um
odesign.modeler.model_units = 'um'

# import dxf file
layer_list = odesign.get_dxf_layers(dxf_path_name)
odesign.import_dxf(file_path=dxf_path_name, layers=layer_list)

odesign.save_project(project_path_name)
odesign.release_desktop(close_projects=True, close_desktop=False)
```

2. I try to import the dxf file manually and export a new dxf file. Then, this new dxf file can be imported by PyAEDT.
Seems like PyAEDT can not decode the original dxf file.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3   
anyio==4.4.0                
argon2-cffi==23.1.0         
argon2-cffi-bindings==21.2.0
arrow==1.3.0                
asttokens==2.4.1            
async-lru==2.0.4            
attrs==24.2.0
babel==2.16.0
beautifulsoup4==4.12.3
bleach==6.1.0
certifi==2024.8.30
cffi==1.17.0
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.3.0
cycler==0.12.1
debugpy==1.8.5
decorator==5.1.1
defusedxml==0.7.1
exceptiongroup==1.2.2
executing==2.1.0
fastjsonschema==2.20.0
fonttools==4.53.1
fpdf2==2.7.9
fqdn==1.5.1
h11==0.14.0
httpcore==1.0.5
httpx==0.27.2
idna==3.8
ipykernel==6.29.5
ipython==8.27.0
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.4
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyter_server==2.14.2
jupyter_server_terminals==0.5.3
jupyterlab==4.2.5
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
kiwisolver==1.4.7
MarkupSafe==2.1.5
matplotlib==3.9.2
matplotlib-inline==0.1.7
mistune==3.0.2
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
nest-asyncio==1.6.0
notebook_shim==0.2.4
numpy==1.26.4
overrides==7.7.0
packaging==24.1
pandas==2.2.2
pandocfilters==1.5.1
parso==0.8.4
pillow==10.4.0
platformdirs==4.2.2
plumbum==1.8.3
pooch==1.8.2
prometheus_client==0.20.0
prompt_toolkit==3.0.47
psutil==6.0.0
pure_eval==0.2.3
pyaedt==0.10.0
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.26.0
Pygments==2.18.0
pyparsing==3.1.4
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.44.1
pywin32==306
uri-template==1.3.0
urllib3==2.2.2
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.8.0
webencodings==0.5.1
websocket-client==1.8.0

gmalinve, 2024-09-10:
Hi @akselchang,

Have you tried to import the dxf manually first? because I read in the error message that AEDT can't decode byte 0xe9 at a certain position.

gmalinve, 2024-09-11:
Hi @akselchang,

Would it be possible for you to share the failing dxf? or a dummy dxf that fails during import? 
Because this method works in our unit test.
This will help us understand the matter.

Thanks

Giulia

akselchang, 2024-09-11:
Hi @gmalinve ,

Sure, please refer the following link:
https://drive.google.com/file/d/1MBmFHkRfQfhN7-R5seFvy2A7_h0WFSWX/view?usp=drivesdk

Thanks for your checking.

gmalinve, 2024-09-11:
Hi @akselchang,

I tried both manually and with PyAEDT and got the same error: 
[image]

akselchang, 2024-09-12:
Hi @gmalinve,

That' weird. I can import this dxf manually with 2024R1/2023R1/2022R2 AEDT.
[image]
[image]
[image]

akselchang, 2024-09-20:
Hi @gmalinve,

I found that if I remove all of the block settings in the dxf drawing, then I can import it by PyAEDT.

Thanks

Aksel

gmalinve, 2024-09-23:
Hi @akselchang,

Sorry for the delay.
Thanks for the feedback, I'll then proceed to close this issue.

Regards,

Giulia

closed 2024-09-23 by gmalinve