ansys/pyaedt #5158: Issues running 25.1 with PyAedt

jsalant22, 2024-09-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am unable to run the unittests with 2025.1. I've tried tests from Circuits and EMIT.

### Steps To Reproduce

Set default_version="2025.1" in conftest.py. Try to run one of the unit tests

Looks like an issue with oeditor not being set properly:

```
test setup failed
add_app = <function add_app.<locals>._method at 0x000001A521985360>

    @pytest.fixture(scope="class")
    def aedtapp(add_app):
        app = add_app(application=Circuit, subfolder=test_subfolder)
>       app.modeler.schematic_units = "mil"

test_21_Circuit.py:51: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
..\src\ansys\aedt\core\application\analysis_nexxim.py:228: in modeler
    self._modeler = ModelerNexxim(self)
..\src\ansys\aedt\core\modeler\schematic.py:492: in __init__
    self._schematic = NexximComponents(self)
..\src\ansys\aedt\core\modeler\circuits\primitives_nexxim.py:104: in __init__
    CircuitComponents.__init__(self, modeler)
..\src\ansys\aedt\core\modeler\circuits\primitives_circuit.py:85: in __init__
    self.refresh_all_ids()
..\src\ansys\aedt\core\generic\general_methods.py:267: in wrapper
    return raise_exception_or_return_false(e)
..\src\ansys\aedt\core\generic\general_methods.py:232: in raise_exception_or_return_false
    raise e
..\src\ansys\aedt\core\generic\general_methods.py:246: in wrapper
    out = user_function(*args, **kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <ansys.aedt.core.modeler.circuits.primitives_nexxim.NexximComponents object at 0x000001A520195EA0>

    @pyaedt_function_handler()
    def refresh_all_ids(self):
        """Refresh all IDs and return the number of components.
    
        References
        ----------
    
        >>> oEditor.GetAllElements()
        """
>       obj = self.oeditor.GetAllElements()
E       AttributeError: 'NoneType' object has no attribute 'GetAllElements'

..\src\ansys\aedt\core\modeler\circuits\primitives_circuit.py:1045: AttributeError
PyAEDT INFO: Desktop has been released and closed.

Process finished with exit code 1
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

agi.stk12 @ file:///C:/Program%20Files/AGI/STK%2012/bin/AgPythonAPI/agi.stk12-12.6.0-py3-none-any.whl
aiohttp==3.9.5
aiosignal==1.3.1
annotated-types==0.6.0
ansys-ansys-wireless @ file:///C:/Users/jsalant/Downloads/wireless%206/wireless/ansys_ansys_wireless-0.1-py3-none-any.whl#sha256=0d437482fb7175b3a2b58dba6872920021d72878b64dfe23edabc1a74f670cd7
ansys-pythonnet==3.1.0rc3
ansys-templates==1.8.0
argcomplete==3.2.1
arrow==1.3.0
astroid==2.15.6
asttokens==2.2.1
async-timeout==4.0.3
attrs==22.2.0
backcall==0.2.0
binaryornot==0.4.4
boto3==1.34.126
botocore==1.34.126
certifi==2022.12.7
cffi==1.15.1
cftime==1.6.3
chardet==5.2.0
charset-normalizer==3.1.0
click==8.1.3
click-plugins==1.1.1
cligj==0.7.2
cloudpickle==3.0.0
clr-loader==0.2.6
collector==0.4.6
colorama==0.4.6
comm==0.1.3
contourpy==1.0.7
cookiecutter==2.2.3
coverage==7.4.3
cycler==0.11.0
dask==2024.5.2
debugpy==1.6.7
decorator==5.1.1
defusedxml==0.7.1
dill==0.3.7
distributed==2024.5.2
docutils==0.16
drjit==0.4.4
et-xmlfile==1.1.0
exceptiongroup==1.1.1
executing==1.2.0
ffmpeg-python==0.2.0
filelock==3.15.1
Fiona==1.9.3
fonttools==4.39.0
fpdf2==2.7.8
frozenlist==1.4.1
fsspec==2024.6.0
future==1.0.0
geopandas==0.12.2
getmac==0.9.5
heyvi==0.2.28
idna==3.4
imageio==2.34.1
imageio-ffmpeg==0.4.9
importlib-metadata==7.0.1
iniconfig==2.0.0
intel-openmp==2021.4.0
ipykernel==6.22.0
ipython==8.13.2
ipywidgets==8.0.6
isort==5.12.0
jedi==0.18.2
Jinja2==3.1.2
jmespath==1.0.1
jsonschema==4.21.1
jsonschema-specifications==2023.12.1
jupyter_client==8.2.0
jupyter_core==5.3.0
jupyterlab-widgets==3.0.7
kiwisolver==1.4.4
lazy-object-proxy==1.9.0
lazy_loader==0.4
lightning-utilities==0.11.2
locket==1.0.0
MarkupSafe==2.1.3
matplotlib==3.7.5
matplotlib-inline==0.1.6
mccabe==0.7.0
mitsuba==3.5.0
mkl==2021.4.0
mpmath==1.3.0
msgpack==1.0.8
multidict==6.0.5
munch==2.5.0
mypy==1.4.0
mypy-extensions==1.0.0
nest-asyncio==1.5.6
netCDF4==1.6.5
networkx==3.1
numpy==1.24.4
opencv-python==4.9.0.80
openpyxl==3.1.5
osmnx==1.3.0
packaging==23.0
pandas==1.5.3
parso==0.8.3
partd==1.4.2
pickleshare==0.7.5
Pillow==9.4.0
pipx==1.4.1
platformdirs==3.1.1
plotly==5.14.1
pluggy==1.0.0
plumbum==1.8.1
pooch==1.7.0
prompt-toolkit==3.0.38
psutil==5.9.4
pure-eval==0.2.2
-e git+https://github.com/ansys/pyaedt.git@8560e9c5d3c581c51309f83c66c41dbc9d11aec1#egg=pyaedt
pyargus==1.1.post1
pybind11-stubgen==0.16.2
pycollada==0.8
pycollector==0.4.6
pycparser==2.21
pydantic==2.7.1
pydantic_core==2.18.2
pydevd-pycharm==242.16677.12
pyedb==0.26.0
pygenstub==2.0.1
Pygments==2.15.1
pylint==2.17.5
pyparsing==3.0.9
pyproj==3.5.0
PyQt5==5.15.6
PyQt5-Qt5==5.15.2
PyQt5-sip==12.12.1
PySide2==5.15.2.1
PySide6==6.4.2
PySide6-Addons==6.4.2
PySide6-Essentials==6.4.2
pytest==7.2.2
pytest-cov==4.1.0
python-dateutil==2.8.2
python-slugify==8.0.1
pythonnet==3.0.2
pytomlpp==1.0.13
pytorch-lightning==2.3.0
pytz==2022.7.1
pyvista==0.43.10
pywin32==305
PyYAML==6.0.1
pyzmq==25.0.2
referencing==0.33.0
requests==2.28.2
rpds-py==0.18.0
rpyc==6.0.0
Rtree==1.2.0
s3transfer==0.10.1
scikit-image==0.23.2
scikit-rf==0.32.0
scipy==1.10.1
scooby==0.7.1
shapely==2.0.4
shiboken2==5.15.2.1
shiboken6==6.4.2
six==1.16.0
snakeviz==2.2.0
sortedcontainers==2.4.0
SRTM.py==0.3.7
stack-data==0.6.2
sympy==1.12.1
tbb==2021.12.0
tblib==3.0.0
tenacity==8.2.2
text-unidecode==1.3
tifffile==2024.4.18
toml==0.10.2
tomli==2.0.1
tomlkit==0.12.1
toolz==0.12.1
torch==2.3.1
torchmetrics==1.4.0.post0
torchvision==0.18.1
tornado==6.3.1
tqdm==4.65.0
traitlets==5.9.0
types-python-dateutil==2.8.19.14
typing==3.7.4.3
typing_extensions==4.12.2
ujson==5.10.0
urllib3==1.26.15
userpath==1.9.1
utm==0.7.0
vipy==1.14.4
vtk==9.2.6
wcwidth==0.2.6
widgetsnbextension==4.0.7
wrapt==1.15.0
yarl==1.9.4
zict==3.0.0
zipp==3.17.0

SMoraisAnsys, 2024-09-16:
Hi @jsalant22, thanks for opening this issue. AFAIK we are not yet trying 25.1 so it's great that you mentioned this problem.
By any change, do you know what changed in AEDT 25.1 which could explain this strange behavior ? AFAIK, we never had a problem at such level before.

Samuelopez-ansys, 2025-01-13:
Hi @jsalant22 ,

Could you let me know if this issue has been solved?

hucares, 2025-06-23:
I want to echo that I am not able to start PyAEDT on 2025.1