ansys/pyaedt #5180: REFACTOR: Leverage pathlib instead of os.path

SMoraisAnsys, 2024-09-18:
We have deprecated IronPython so we should be able to use `pathlib` instead of working with `os.path`.

Updating this will improve readability and follow the push from the Python ecosystem.
This should be a good entry point for new contributors.

jvela018, 2024-10-28:
@gmalinve you can assign this issue to me. Thanks

jvela018, 2025-02-18:
# List of files containing os.path / Completed refactoring
- [x] maxwell.py
- [ ]  twinbuilder.py
- [ ] aedt_logger.py
- [ ] circuit.py
- [ ] circuit_netlist.py
- [ ] common_rpc.py
- [ ] desktop.py
- [ ] downloads.py
- [ ] configurations.py
- [ ] filesystem.py
- [ ] general_methods.py
- [ ] grpc_plugin_dll_class.py
- [ ] ibis_reader.py
- [ ] load_aedt_file.py
- [ ] settings.py
- [ ] hfss.py
- [ ] hfss3dlayout.py
- [ ] icepak.py
- [ ] maxwellcircuit.py
- [ ] object_3d.py
- [ ] modeler_pcb.py
- [ ] q3d.py
- [ ] local_server.py
- [ ] rpyc_services.py
- [ ] add more as needed...!

# Refactoring mapping

- To map from os.path to pathlib (PurePath or Path) follow the table at the end of the pathlib documentation:
https://docs.python.org/3/library/pathlib.html

- Most of the mapping is straight forward but some of them will require some extra effort or completely changing the syntax. For example getting the extension of a file will be done through stem instead of splitting the path and extracting the extension as the argument "1" of a list
    - Old: os.path.splitext(output_file)[1]
    - New: Path(output_file).suffix 

- If you refactor any of the files in the list - submitted a PR and merged it into main, please mark it as completed. 
- If you are uncertain on how to map the functions from one module to the other feel free to ping me.
- Be mindful in the use of Path vs PurePath

closed 2025-08-05 by jvela018