ansys/pyaedt #5194: REFACTOR: Leverage type hint and add checking

SMoraisAnsys, 2024-09-23:
We have deprecated IronPython so we should be able to use type hints ! This will help us to improve code clarity, catch potential bugs early, **enhance IDE support** (this one is particularly great), simplify maintenance, ...

For example

```python

def retrieve_property(self, property_name: str) -> str :
    pass
```

Related to https://github.com/ansys/pyaedt/issues/5152

MaxJPRey, 2024-09-23:
It is for sure a huge work load, but definitely worth it, especially as you said for development in IDE by our users.

vatsalrathod16, 2024-09-24:
I can work on this issue

SMoraisAnsys, 2024-09-24:
> I can work on this issue

Hey @vatsalrathod16, thanks for helping us, any contribution is welcome !
This is a **HUGE** task so I would advise to only work on a subset of the files. Feel free to pick whichever you like 👍