ansys/pyaedt #5207: (AO) 4 terminal Sparamater model pin oder assignment in edb.configuration.load

aohta2021, 2024-09-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I could not control correct pin oder assigment on attached Sparamater file.
Refdes "DL1" is assigned Sparamater model(TCM0403S_350_2P_T210.s4p)

SIwave GUI can confirme Sparamater model(TCM0403S_350_2P_T210.s4p) assignment after imported its EDB.
[To_TEST_925.zip](https://github.com/user-attachments/files/17127975/To_TEST_925.zip)

### Steps To Reproduce

TEST batch script is
1. Clean.bat
2. run_EDB.bat
Check refdes "DL1" Pin Order

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

(venv310) C:\Users\aohta\OneDrive - ANSYS, Inc\Documents\2024_BMT\Sony\Auto\To_TEST_925>python -m pip freeze
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==24.2.0
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.3.0
cycler==0.12.1
defusedxml==0.7.1
et-xmlfile==1.1.0
fast_simplification==0.1.7
fiona==1.10.1
fonttools==4.54.1
fpdf2==2.7.9
geopandas==0.14.4
idna==3.10
imageio==2.35.1
joblib==1.4.2
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.7
markdown-it-py==3.0.0
matplotlib==3.9.2
mdurl==0.1.2
meshio==5.3.5
networkx==3.3
numpy==1.26.4
openpyxl==3.1.5
osmnx==1.9.4
packaging==24.1
pandas==2.2.3
pillow==10.4.0
platformdirs==4.3.6
plumbum==1.8.3
pooch==1.8.2
psutil==6.0.0
pyaedt==0.10.2
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.28.0
Pygments==2.18.0
pyparsing==3.1.4
pyproj==3.6.1
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.2
pyvista==0.44.1
pywin32==306
PyYAML==6.0.2
referencing==0.35.1
requests==2.32.3
rich==13.8.1
rpds-py==0.20.0
rpyc==6.0.0
Rtree==1.3.0
scikit-learn==1.5.2
scikit-rf==1.3.0
scipy==1.14.1
scooby==0.10.0
shapely==2.0.6
six==1.16.0
SRTM.py==0.3.7
threadpoolctl==3.5.0
toml==0.10.2
typing_extensions==4.12.2
tzdata==2024.2
urllib3==2.2.3
utm==0.7.0
vtk==9.3.1

aohta2021, 2024-09-30:
The case of attached SPICE Pinorder is avalable as bellow.
	"spice_models": [
        {
            "name": "TCM0403S_350_2P_T210",
            "component_definition": "DLW31SN601",
	    "component_type": "Inductor",
            "file_path": "./tcm0403s_350_2p_t210_s.mod",
            "apply_to_all": false,
            "components": ["DL1"],
	    "terminal_pairs": [["n1", "A1"], ["n2", "B2"], ["n3", "A2"], ["n4", "B1"]],
            "reference_net": "VSS"
        }
    ]
But  "component_type": "Inductor" could not changed.
Let me know its change correct format.
And also user want to set by "terminal_pairs" for S paramater model.

svandenb-dev, 2024-10-20:
Moving to PyEDB.
@hui-zhou-a this issue is fixed ?